openwrt/luci #6871: luci wishlist

systemcrash, 2024-01-26:
Is the following possible?

`form.DynamicList` have the ability to 'edit' an existing entry (as opposed to just 'delete' with the red X), useful for single character modifications

Big flexibility improvement.

jow-, 2024-02-18:
- What should happen if the edited value ends up being the same as an existing value?
- What should happen if the edited value is emptied out?
- How should the "edit mode" be triggered?
  - How should it be triggered with keyboard navigation?
- How should one leave edit mode?
   - Should there be a difference between "applying" the changed value (e.g. Enter) and "cancelling" the edit (e.g. Esc, Arrow up/down, Tab out)?

systemcrash, 2024-02-18:
> * What should happen if the edited value ends up being the same as an existing value?

I think at least error checks and verify functions should run. This also has the added effect of catching problems that have slipped into config via other vectors. 

> * What should happen if the edited value is emptied out?

Lots. The above and that entry can at least be culled from a list. 

> * How should the "edit mode" be triggered?

Click into or tab into the field. It should be simple and such functionality be optionally also "hinted" by some interface element or colouring which is intuitive to new or accsustomed alike. Like a small green triangle in one of the upper corners of that field, or some glyph of pen and scribble writing with alpha transparency making it subtly visible.  

>   * How should it be triggered with keyboard navigation?

Tab is good. Any sane interface shall be tabbable. I think tab and then an additional key press to unlock the field is already too many actions. People who have cats running across their keyboards is a first world problem we can ignore for now. 

> * How should one leave edit mode?

ESC is the universally understood panic button. 

>   
>   * Should there be a difference between "applying" the changed value (e.g. Enter) and "cancelling" the edit (e.g. Esc, Arrow up/down, Tab out)?

ESC should be the only way to escape. 😅

Tab away, enter, as suggested, make operations simple. No clicking extra things otherwise you get stuck in click hell if you want to tab forward and then back to the same field. Simplify as much as possible. But not more. 👌

bracklanna, 2024-04-21:
MAJOR wishlist item:

The ability to schedule ON/OFF times on a PER-SSID basis.

In a case where there are multiple SSIDs, i.e. "Guest", "Residents", "Staff" etc., it is important to be able to specify WHICH SSIDs stay active and which do not, according to different schedules.

systemcrash, 2024-04-21:
> MAJOR wishlist item:
> 
> The ability to schedule ON/OFF times on a PER-SSID basis.
> 
> In a case where there are multiple SSIDs, i.e. "Guest", "Residents", "Staff" etc., it is important to be able to specify WHICH SSIDs stay active and which do not, according to different schedules.

So... I could just enable or disable firewall rules which block traffic to said SSIDs and achieve the same result, no?

In any case, this isn't luci functionality - it's lower down in the stack. Luci is (generally) only the interface to functionality.

closed 2025-03-04 by hnyman