openwrt/luci #6878: „Weblate“ locked (again), errors. (CONFLICT (content): Merge conflict in applications/luci-app-firewall/po/lt/firewall.po)

dziugas1959, 2024-01-31:
As I already mentioned in „Help translate the web interface via weblate“
[image]
[image]

dziugas1959, 2024-01-31:
P.S all languages are affected.

hnyman, 2024-01-31:
Yeah, I am pretty sure that it is related to commits made directly to language files by @systemcrash a few hours ago.
I haven't yet figure out which of the recent commits it was.
(Firewall, network and luci-base are somewhat intertwined)

Weblate is annoyingly picky about direct git changes to files that are included in its open PRs.
As best practice to avoid conflicts, I have learned to first merge the open Weblate PRs, and only then edit / merge other language file changes.

hnyman, 2024-01-31:
Fixed with b6e968b

This was a pretty peculiar failure. 
No clear direct git commits touching firewall or statistics components, except weblate's own PR commits. 

Looks like a race/concurrency problem, where somebody have just translated one or more strings and weblate has launched an periodic commit process and it simultaneously receives a trigger of upstream repository change (direct git commit to LuCI repo).

The first failed/conflicting strings are translated here:
[image]

When a few hours late @systemcrash makes edit via weblate interface, it then suddenly notices a conflict. Based on the edits needed, the conflict was related to weblate changes made by @dziugas1959 a few hours earlier via weblate itself.

[image]

So, weblate was in conflict with changes made via itself. :-(
First time that we have seen that, I think.

closed 2024-01-31 by hnyman

systemcrash, 2024-01-31:
😅