openwrt/luci #6885: luci-app-dockerman missing main menu items

Loafter, 2024-02-03:
luci-app-dockerman broken it's show only configuratation button, and apear only after reinstall it

for repeat bug. 

1. install luci-app-dockerman with depends
2. check that after installation full menu available
[image]
3. reboot router
Now full menus is missing available only config
[image]
4.  remove luci-app-dockerman package
5.  install luci-app-dockerman package
6.  it's apear again
7. [image]

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.2'
DISTRIB_REVISION='r23630-842932a63d'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 23.05.2 r23630-842932a63d / LuCI openwrt-23.05 branch git-24.006.68745-9128656'

systemcrash, 2024-02-04:
Anyone with lua experience who can look at this? dockerman needs porting to js anyway.

stokito, 2024-02-27:
Looks similar to #6280

@systemcrash before converting the dockerman to JS we should check and grab changes from upstream https://github.com/lisaac/luci-app-dockerman.

We may convert easily some parts but for other we'll need to call a lua script which is a shim for a Docker REST API. That may be resolved later, this is not critical to still have some non UI parts in Lua.

Basically I started from this task but there wasn't a good solution and I paused and switched to other tasks.

So if you'll have a desire to convert the dockerman to JS please let me know and I'll join. I have it somewhere in a middle of my TODO.

systemcrash, 2024-02-27:
Is ucode an option?

stokito, 2024-02-27:
It should be possible to make in a plain shell. Still, that's not blocking to convert the UI to JS.

tignioj, 2024-02-28:
Same issue.

dannil, 2024-03-17:
When fixing https://github.com/openwrt/luci/pull/6998 I discovered that I was able to make the menu behave as before by installing `dockerd` on my host. I'm definitely not fluent in Lua but it looks like the code which configures the menus exits early if it can't contact either the remote or local socket. This may also be related to https://github.com/openwrt/luci/pull/6985 however, if my suspicion is correct, the solution would probably be to make the menus behave appropriately even if a socket connection can't be made.

https://github.com/openwrt/luci/blob/f2a49104c4839cc61c0c6a32cf0b616ca7b7d667/applications/luci-app-dockerman/luasrc/controller/dockerman.lua#L18-L34

JKSTAFF, 2024-03-26:
My guess is that the order of service startup and page loading may be causing the display error.  
Here is my  reproduce. After a normal system boot, when I could access the incomplete luci-dockerman, type docker command in console reported this:  
```
Cannot connect to the Docker daemon at unix:///var/run/docker.sock. Is the docker daemon running?
```
So I waited for all containers to autostart, and then docker command returned to usable. At this time, I clear the luci cache under the tmp and now I can access the missing menu again through the subpath.

flhofer, 2024-03-29:
Hi,
you can simply change the startup order by changing the priority in `/etc/rc.d/`

If, for example, set it to 45, before uhttpd at 50.
`mv S99dockerd S45dockerd`

Now it is visible again. Note that the priority in the web interface does not update. That might be because its generated at trigger only.

JKSTAFF, 2024-03-29:
> Hi, you can simply change the startup order by changing the priority in `/etc/rc.d/`

Thank you for reply but I already tried that way (in my build it's nginx instead of httpd) and cause the same result

flhofer, 2024-03-29:
Hi @JKSTAFF 

> I also changed dockerd start priority in /etc/init.d
What do you mean by that?

Don't confuse `init.d` with `rc.d`. The latter sets the order and contains only symlinks to the scripts in `rc.d`. The `S` or `K` stands for start or kill, and the 2-digit number is the order, where 99 is last. If you use nginx as server, you might check what order that is and change the number accordingly. 

Let me know if you need further explanation

JKSTAFF, 2024-03-29:
Hi @flhofer Thank you for your patience and explanation. In my build modifying dockerd lower than 60 takes precedence over nginx(S60).  
>I also changed dockerd start priority in /etc/init.d  

I mean I also modified `/etc/init.d/dockerd` like this:    
```
#!/bin/sh /etc/rc.common

USE_PROCD=1
START=99 → 50   ## I changed here

extra_command "uciadd" "<interface> <device> <zone> Add docker bridge configuration to network and firewall uci config"
..............
```

Loafter, 2024-03-29:
Yep.
Guys look like this bug still appears. I have two x86 OpenWrt Host with latest 23.05.3 version and latest 24 luci-docker plugin. and one of this host  display full menu items and second with missing menu items, difference only with count of container, with displayed have only one container and onother about 7 containers

flhofer, 2024-03-29:
@Loafter @JKSTAFF, so it looks like if you have many containers to run at the start, the daemon is not responsive right away. In my case, it works as I have only two smaller containers. Maybe we need some autostart delay for dockerd's containers.

Rategh, 2024-06-16:
I'm experiencing the same issue  on a Raspberry Pi. I only have one container running.

zxn10, 2024-08-18:
I had the same issue. The dropdown menu sometimes appears and sometimes doesn't.

systemcrash, 2024-10-08:
Closed by #6985

closed 2024-10-08 by systemcrash

JKSTAFF, 2024-12-07:
missing main menu items still exist with dockerd and docker, docker-compose and luci-lib-docker.   
I compiled firmware with complete dockerman dependencies on both 23.05.5 and 24.10-rc2,  without any connecting problem to docker website, I guess the problem still comes from dockerd failing to start automatically after booting. I can access the missing menu options in the address bar only after manually restart dockerd service.  
If I don't do this, the mandatory access options address will be like this on browser:  
```
No page is registered at '/admin/docker/images'. If this url belongs to an extension, make sure it is properly installed. If the extension was recently installed, try removing the /tmp/luci-indexcache file.
```

mrpops2ko, 2025-07-11:
having this issue with the latest version of openwrt too

removing the `luci-index` file in `/tmp/` ``` /tmp/luci-indexcache``` and logging in and out fixed it for me

running on a powerful x86 machine, not sure why this is happening as the fix doesn't persist across reboots

following as described in this thread and venturing to ```/etc/init.d/dockerd``` and changing the boot order seems to have fixed it. changing the `START=99` to `START=50` 

i've tried across 5 reboots now and everything works again.