openwrt/luci #6890: modules/luci-base/luasrc/sys.lua: Process list not showing all processes depending on memory size

PRIMETSS, 2024-02-04:
## Actual behavior:
Some processes of size > 200Mb show in `ps` but not in Luci Web page 'Processes'

## Expected behavior:
ALL processes running should be displayed in page list

This is makes installing large memory use services confusing, as when started they show as running in CLI ps but not in Luci/Processes page
[image]
[image]
[image]

## Steps to reproduce:
> Install an App that consumes a few > hundred Mb or ram. [Probably needs to be custom App and most packages are small, we used a dotnet custom App called "RMS"]
> From CLI view processes 'ps'
> From OpenWRT/Luci web interface open menu /status/processes
> Compare 'ps' output process list with /status/processes list and notice any process reporting 'XXXXm' VSZ size now displays in list but any larger processes of size 'XXXXg'

Previous issue about same issue was raised and 'fixed' but the issue will re-occur if memory process is consuming goes above 'XXXXm' and minimized to 'XXXXg' this is because /status/processes uses the `top` BusyBox CLI command to list processes and then uses a Regex filter to parse the output from `top`

Process reported by `ps`
[image]
Process reported by `top`
[image]
Notice the columns 'VSZ' & %VSZ are concatenated

This is the code in Luci where Regex was fixed for VSZ 'reduced' size to XXXm but it breaks again if an even larger App is running that uses more memory and memory is reported as 'XXXg'

The fault causing this behavior is actually in BusyBox, because its column sizes and the function that does the memory scalar 'XXXm' 'XXXg' reduction doesn't increase the column size and no space is placed between the two column, ccausing the RegEx espression to concatenate the two columns together `+(%d+m?) +(%d+%%)`

The fix for this is to replace with `+(%d+(m|g)?)` which handles the 'm' & 'g' scalar, BUT that does not resolve the root cause of the issue in BusyBox that is caused by a bug in the 'Top' command.
[image]

This I believe is caused here

Top code:
[https://github.com/mirror/busybox/blob/2d4a3d9e6c1493a9520b907e07a41aca90cdfd94/procps/top.c#L710](https://github.com/mirror/busybox/blob/2d4a3d9e6c1493a9520b907e07a41aca90cdfd94/procps/top.c#L710)

So the fault first needs to get fixed in BusyBox then above fix for OpenWRRTLuci.
I'm working on a PR for BusyBox 'Top' fix.
My suggested approach is to get that PR accepted so it can be also fixed in OpenWRT/Luci with anther related PR.
Any comments or other suggestions welcome! 
TY!

hnyman, 2024-02-04:
Please note that there is no modules/luci-base/luasrc/sys.lua in the current LuCI master and stable 23.05. It only exists in old 22.03 (and older) branches. 22.03 is already at only security maintenance status with no active development.

Corresponding JavaScript based code is in master
https://github.com/openwrt/luci/blob/master/modules/luci-base/ucode/sys.uc#L11

The fix should primarily be done for master and then backported to 23.05.

The m/g fix sounds essential for LuCI, but the 14224% looks like something in busybox itself, like you say. The concatenation is a byproduct of exceeding 100%.

Ps. Do you really have a process consuming 258GB memory in OpenWrt context ????

PRIMETSS, 2024-02-04:
Thanks not sure where I got that module name from, search of repo I take it.
Pretty sure this is the correct module for displaying Processes:
[luci/modules/luci-mod-status](https://github.com/openwrt/luci/blob/1e1c6b853157febd3d93cdcbaa059a4f1a80b33b/modules/luci-mod-status/htdocs/luci-static/resources/view/status/processes.js#L18)

Re the odd VSZ 10'000% columns, it's never made sense to me what it is trying to convey.
Have located the code that produces these columns in BusyBox but again not real clear on approach they using for column widths, seems fairly complicated for some reason...
Spent today getting a debug'able local Repo of the BusyBox repo to have a better look. As I was not having much luck understanding the code with just a compile/run approach.

hnyman, 2024-02-04:
To me the busybox column-width mishap is just a harmless byproduct of the actual error of one process claiming to use 14224% = 142x of the total memory amount (which should include that one process??). The 258g is so absurd  that I wonder if there is some 32bit overflow error somewhere. You should probably look at the actual kernel data from /proc and try to understand, which are are original numbers that busybox tries to understand, calculate and show.

systemcrash, 2024-02-04:
Two columns getting conflated is a sign that some screen-scraping is going on. It would be more effective to `cat` the contents of the `/proc/{id}/` files e.g. `/proc/{id}/status`. I thought that's what `ps` does anyway.

258g might be wild, but some Go binaries might be able to do this :D

PRIMETSS, 2024-02-05:
Located the issue and was able to build a custom Busybox with the 'fix' and works as expected.
Problem occurs as OpenWRT uses the Busybox 'Top' command to produce the process list.
`busybox top` has an issue with the way it builds each row of processes it displays in the Luci/Processes page.
The fault is caused because two columns VSZ & %VSZ merge together if the value is >5chars in length. %VSZ concatenates into VSZ removing the space between columns. This then makes the Regex match 'fault' and a column is missed causing the process to get dropped from the process list.

The fault is here `"\n" "%s %s  %.5s" FMT`
'FMT is a c++ Macro `# define FMT "%4u%%"` which when the compiler substitutes it becomes `%.5s%4u%%`
Because there is no space in either ` "%s %s  %.5s"` or  `FMT` the FMT macro under certain variable char length a space does not exist.
[https://github.com/mirror/busybox/blob/2d4a3d9e6c1493a9520b907e07a41aca90cdfd94/procps/top.c#L737](https://github.com/mirror/busybox/blob/2d4a3d9e6c1493a9520b907e07a41aca90cdfd94/procps/top.c#L737)

Fix is just to pad a space between the two variables
`"\n" "%s %s  %.5s "` FMT`

OpenWRT is using a older version of Busybox and this table in docs has not been updated for some time [https://openwrt.org/docs/techref/busybox](https://openwrt.org/docs/techref/busybox)

Currently investigating if this issue has been fixed in current stable version, but at moment don't seem to be able to replicate this issue on current stable version. 1.36

systemcrash, 2024-02-06:
Good find. In this case, the bug is likely in openwrt/openwrt

hnyman, 2024-02-06:
> OpenWRT is using a older version of Busybox

Current OpenWrt main/master is using the newest busybox 1.36.1

jow-, 2024-02-18:
Closing this as it will be fixed OPenWrt side through a Busbox update.

closed 2024-02-18 by jow-