openwrt/luci #6892: [luci-base] issues with drop down menus

xize, 2024-02-06:
## Steps to reproduce:
1. first I start Brave browser version 1.62.156 Chromium: 121.0.6167.139 (64bit Windows 11)
2. then I go in luci -> network -> interfaces
3. create around 10 interfaces
4. when I try to change the device of the lan interface and move my cursor to select a item and scroll, the main scrollbar conflicts with the scrollbar on the browsers window resulting in very glitchy behaviour, theres also a similar issue for the firewall zones.

my screen resolution is 1920 * 1080.

currently I use the following snapshot version (not gl-inets firmware):
```
root@MT6000:~# cat /etc/os-release 
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r25024-5e395f0c24"
OPENWRT_BOARD="mediatek/filogic"
OPENWRT_ARCH="aarch64_cortex-a53"
OPENWRT_TAINTS=""
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r25024-5e395f0c24"
```
^ this version is not from auc.

and ubus board show this info:
```
root@MT6000:~# ubus call system board
{
        "kernel": "5.15.148",
        "hostname": "MT6000",
        "system": "ARMv8 Processor rev 4",
        "model": "GL.iNet GL-MT6000",
        "board_name": "glinet,gl-mt6000",
        "rootfs_type": "squashfs",
        "release": {
                "distribution": "OpenWrt",
                "version": "SNAPSHOT",
                "revision": "r25024-5e395f0c24",
                "target": "mediatek/filogic",
                "description": "OpenWrt SNAPSHOT r25024-5e395f0c24"
        }
}
```

I do know this issue is also a little before this version, actually i observed it first on version r24993-32b4498c15, please let me know if I need to give any other information thanks 👍

jow-, 2024-02-18:
You mean that the main page scrolls beneath the dropdown while you scroll through the drop down? Do you use arrow keys? Tabbing? Mouse wheel?

xize, 2024-02-19:
this happened only when I clicked on such drop down element, when I move my cursor and hovering inside the element then it was conflicting with the main scrollbar, it looked similar as that space was being hold but then very erratic, it worked better with arrow keys.

strangely enough I checked this again and it seems to be no more present so maybe it was a issue on my side either client side through cache or maybe a sticky key.

but it could also be related to this issue: https://github.com/openwrt/luci/issues/6903 although I did not updated my firmware but it seems to make sense if i follow the progress of it.