openwrt/luci #6902: luci-mod-network: wireless.js: Key input disappears for PSK/PSK2 client modes

mkrkn, 2024-02-09:
## Steps to reproduce:
1. go to: Network → Wireless → Edit on any wireless interface
2. Interface Configuration -> General Setup: Set Mode to Client or Client (WDS)
3. Interface Configuration -> Wireless Security: Set Encryption to WPA2-PSK, WPA-PSK (or any non-WPA3 PSK option)
4. Observe no Key input to define pre-shared key

## Actual behavior:
Any non-WPA3 encryption mode using pre-shared key for authentication will hide the Key field. Access point modes work fine with any encryption. Client modes with WPA3-SAE also work.

## Expected behavior:
Key field should appear for all pre-shared key modes as it is required

## Additional Information:
Bisected LuCI repository and identified 75a2fd233baabf6fa5bafcb13ccc6d7000a4604d as the problem commit. I don't understand exactly how the dependency works, it could be that the Key field inherits the dependencies of the `ppsk` item, which depends on Access point modes. WPA3 dependencies are added afterwards, which would explain why WPA3-SAE allows Key entry.

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r25004+2-b463737826'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r25004+2-b463737826'
DISTRIB_TAINTS='no-all'
```

systemcrash, 2024-02-12:
@mgiganto any comments?

systemcrash, 2024-02-12:
I had a quick play with the GUI and it seems some weird interplay in that rats-nest of dependencies. Setting a dep to PSK and variants exclusively works fine, but adding ppsk: '0' makes it disappear...

jow-, 2024-02-14:
This issue is serious and there's no reaction from the original commit author, please revert this faulty change.

systemcrash, 2024-02-14:
Resolved in 05af14b25bc0fc2262416dcebf3360b11104198b

closed 2024-02-14 by systemcrash