openwrt/luci #6903: Focus problem when entering custom values to drop-downs

jernejs, 2024-02-10:
## Steps to reproduce:
Go to Network → DHCP and DNS → Static Leases tab → click on Add button → open the MAC address drop-down → click the `-- custom --` textbox
(same problem happens in other drop-downs that allow custom input, eg. Network → Interfaces → Edit → Firewall Settings tab → Create / Assign firewall-zone)

## Actual behavior:
If you move the mouse pointer out of the textbox, the textbox loses focus, and you can't type in it any more

## Expected behavior:
Textbox retains focus regardless of the mouse pointer position

## Additional Information:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.2'
DISTRIB_REVISION='r23630-842932a63d'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 23.05.2 r23630-842932a63d'
DISTRIB_TAINTS='no-all'
```
`Powered by LuCI openwrt-23.05 branch (git-23.306.39416-c86c256 / OpenWrt 23.05.2 (r23630-842932a63d)`

The problem happens regardless of selected LuCI theme (tested with Bootstrap, Material and OpenWrt2020).

systemcrash, 2024-02-10:
Your OS?

Seems like an OS or mouse driver feature, bring focus to window under
pointer.

> Message ID: ***@***.***>
>

jernejs, 2024-02-10:
OS is Windows 11 and this happens on both Vivaldi and Firefox, and specifically only affects textboxes in LuCI's drop-downs; other textboxes (even in LuCI) are not affected. I do not have focus follows mouse enabled (I always thought that was a misfeature on X); additionally, after the drop-down textbox loses focus, I have to click inside to get it to focus again, it does not regain focus if I just move my mouse back.

systemcrash, 2024-02-10:
Ah, I see it now. I guess don't move the mouse then :)

This looks like a job for @jow-

commit 3980c192 (2024-02-18): luci-base: ui.js: rework dropdown focus behavior

Drop mouse following focus behavior as it interferes with custom value inputs,
rely on CSS based hover effects instead.

Also slightly improve keyboard navigation by auto-focussing the custom value
input when entering the last dropdown choice via arrow down or tab key, and
by allowing to leave the text input again with the arrow up key.

Fixes: #6903
commit d2f57f77 (2024-03-07): luci-base: ui.js: rework dropdown focus behavior

Drop mouse following focus behavior as it interferes with custom value inputs,
rely on CSS based hover effects instead.

Also slightly improve keyboard navigation by auto-focussing the custom value
input when entering the last dropdown choice via arrow down or tab key, and
by allowing to leave the text input again with the arrow up key.

Fixes: #6903

(cherry picked from commit 3980c192cda6985c2e2d2608e74677cf236005db)
commit f14c04a8 (2024-03-07): luci-base: ui.js: rework dropdown focus behavior

Drop mouse following focus behavior as it interferes with custom value inputs,
rely on CSS based hover effects instead.

Also slightly improve keyboard navigation by auto-focussing the custom value
input when entering the last dropdown choice via arrow down or tab key, and
by allowing to leave the text input again with the arrow up key.

Fixes: #6903

(cherry picked from commit 3980c192cda6985c2e2d2608e74677cf236005db)
closed 2024-02-18 by jow-