openwrt/luci #6913: OpenWRT Luci login page cannot access with openwrt version 23.05.0 with raspberry pi 4

mrusli70, 2024-02-14:
I am using Raspberry pi 4 with version of 23.05.0 after upgrading the packages the luci login page no longer be accesable. 
I even try http://openwrt.lan still it does not work. I have try to reinstall luci and luci-ssl and restart the /etc/init.d/rpcd and /etc/init.d/uhttpd services it still does not seems to work at all. How do i get the login page to solve this problems that i had.

systemcrash, 2024-02-14:
I found that having https enabled really fucks up and dies. Leaving http
only enabled works or restarting the uhttpd daemon for some reason fixes it
after boot up.

Try: /etc/init.d/uhttpd restart

Or disable https.

I have not looked into this more. And this might not be your case.

jow-, 2024-02-14:
Define "does not work". Connection refused? Certificate mismatch? Exception message / stacktrace printed by LuCI?

jow-, 2024-02-18:
Not enough information to work with.

closed 2024-02-18 by jow-