openwrt/luci #6921: [luci-proto-wireguard] "Generate configuration" does not update the QR code and config file according to the values in the form

floatingpurr, 2024-02-16:
## Steps to reproduce:
0. Create a new Wireguard Network Interface
1. Create e new peer
2. click on "Generate configuration..." next to the key "Configuration Export"
3. Change the values in the "Generate configuration" tab

## Actual behavior:
The QR code and the config file (printed out next to the QR code) are not updated

## Expected behavior:
The QR code and the config file (printed out next to the QR code) should be updated

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r25142-a42075435a'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r25142-a42075435a'
DISTRIB_TAINTS=''
```
Package version: `luci-proto-wireguard git-23.338.83313-5b26887`

systemcrash, 2024-02-16:
Perhaps you can propose a fix?

jow-, 2024-02-16:
@systemcrash - this broke with your commit 5b26887c52097dc9364c9b4f4dfb2d6ba2f08818

closed 2024-02-16 by jow-

systemcrash, 2024-02-16:
Snap - I was about to add a different fix. Good catch.

commit e28f9a04 (2024-02-16): luci-proto-wireguard: fix broken DNS option in peer config generator

The config generation code was referencing a not existing `dns_servers`
input field.

Fixes: #6921
Fixes: 5b26887c52 ("Adding a DNS option to the wireguard peer config ...")
commit 7964117f (2024-02-16): luci-proto-wireguard: fix broken DNS option in peer config generator

The config generation code was referencing a not existing `dns_servers`
input field.

Fixes: #6921
Fixes: 5b26887c52 ("Adding a DNS option to the wireguard peer config ...")

(cherry picked from commit e28f9a0491f5f50a70bbefb1720ff8241f456f83)
commit 270b7373 (2024-02-16): luci-proto-wireguard: fix broken DNS option in peer config generator

The config generation code was referencing a not existing `dns_servers`
input field.

Fixes: #6921
Fixes: 5b26887c52 ("Adding a DNS option to the wireguard peer config ...")

(cherry picked from commit e28f9a0491f5f50a70bbefb1720ff8241f456f83)