openwrt/luci #6930: luci-mod-network: missing generate pmk locally breaks WPA3-SAE roaming

maciekczwa, 2024-02-18:
## Steps to reproduce:

1. go to: Network → Wireless → Edit network with WPA2 -PSK/WPA3-SAE Mixed Encryption
2. go to: WLAN roaming tab
3. The Generate PMK locally option is missing
4. Save

## Actual behavior:
1. The ft_psk_generate_local is removed from config.
2. The default of ft_psk_generate_local is 1 when it is not in config which prevents 802.11r from working with WPA3-SAE

## Expected behavior:

```
The Generate PMK locally option should be visible but disabled and set to 0 when using WPA3-SAE.

Or second option is to hide it, but set to 0 when saving changes.

```

## Additional Information:
luci-mod-network version 	git-24.046.07140-9939534

This was probably broken by changes in
https://github.com/openwrt/luci/commit/4c76a91fd7aabdef0a8bc8769727b909e994a628

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.2'
DISTRIB_REVISION='r23630-842932a63d'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 23.05.2 r23630-842932a63d'
DISTRIB_TAINTS=''
```

McGiverGim, 2024-03-06:
I think the problem is the `o.rmempty = false;`. That implies that when it's hidden, it maintains the uci value, it does not remove it.

@jesferman it seems the commit is yours, what do you think?

McGiverGim, 2024-03-06:
I've seen that de "default" value for this option is `1` so removing it will not serve. It needs to be put to zero.

jow-, 2024-03-06:
> The default of ft_psk_generate_local is 1 when it is not in config which prevents 802.11r from working with WPA3-SAE

This does not appear to be the case for OpenWrt main (snapshot): https://github.com/openwrt/openwrt/blob/main/package/network/config/wifi-scripts/files/lib/netifd/hostapd.sh#L927

For 23.05 however the behavior looks wrong:
https://github.com/openwrt/openwrt/blob/openwrt-23.05/package/network/services/hostapd/files/hostapd.sh#L912

The `sae` and `psk-sae` cases should be removed. It seems that there's nothing to fix on the LuCI side but that OpenWrt commit e2f6bfb833a1ba099e1dcf0e569e4ef11c31c391 should be backported to make the implicit default value of `ft_psk_generate_local` dependent on the auth suite.

closed 2024-03-06 by jow-

jesferman, 2024-03-06:
Late to the party but I agree with jow, backporting my commits to 23.05 should fix the issue. And I see that jow has just commited the changes, so everything should be fine.

protectivedad, 2024-07-14:
Maybe I should open a new issue, not sure. But the use of EAP doesn't seem to have been taken into account when making these changes. I am using a radius server with WPA2 and I want to set the R0 and R1 keys. The "Generate PMK locally" option doesn't show up and even worse if I set it in the wireless file it is deleted when I save any changes using the Luci wireless interface. I am using OpenWrt 23.05-SNAPSHOT, r24003-0cdbbd8868

jesferman, 2024-07-14:
If I understand you correctly, you are trying to use WPA2-EAP and you want to use local PMK, but that is not possible. Local PMK is only possible when using PSK, if you use EAP, local PMK can not be used and the option to set it is hidden.

protectivedad, 2024-07-14:
Other way around. I do NOT want to use local PMK. I am setting global R0 and R1. But the option ft_psk_generate_local defaults to 1 and if I set it to 0 it get deleted from the config file when I update any wifi settings through Luci.

protectivedad, 2024-07-14:
Sorry, I just looked into the hostapd configs and saw that it sets them to '0'. So it isn't really defaulting to '1' it is defaulting to a value that makes sense for the key being used. Which means for me that I was creating a problem that doesn't exist.