openwrt/luci #6933: luci-base: Promise traceback

systemcrash, 2024-02-20:
uci.load('wireless') causes a traceback. 

https://github.com/openwrt/luci/blob/c8cddc35302380882bab910679f717d65f715029/modules/luci-base/htdocs/luci-static/resources/network.js#L360

My VM has no 'wireless' devices or such 'wireless' uci sections. 

This load is called, but no value is immediately processed in this function definition. 

I was able to get by without these lines but I am unsure of the consequence of doing so yet without more testing:

https://github.com/openwrt/luci/blob/c8cddc35302380882bab910679f717d65f715029/modules/luci-base/htdocs/luci-static/resources/network.js#L359-L361

All the mentions of `data` - which are the result of promise resolutions - within function scope are here:

https://github.com/openwrt/luci/blob/c8cddc35302380882bab910679f717d65f715029/modules/luci-base/htdocs/luci-static/resources/network.js#L362-L369

so it feels like these load calls can be excised. But maybe they are there for some caching effect.

@jow- ?

Traceback:
```
RPCError: RPC call to uci/get failed with ubus code 4: Resource not found
  at handleCallReply (http://192.0.2.1/luci-static/resources/rpc.js?v=git-23.292.78392-9f66674:15:3)
  at promise callback*parseCallReply (http://192.0.2.1/luci-static/resources/rpc.js?v=git-23.292.78392-9f66674:9:70)
  at promise callback*call (http://192.0.2.1/luci-static/resources/rpc.js?v=git-23.292.78392-9f66674:5:110)
  at declare/</< (http://192.0.2.1/luci-static/resources/rpc.js?v=git-23.292.78392-9f66674:23:320)
  at declare/< (http://192.0.2.1/luci-static/resources/rpc.js?v=git-23.292.78392-9f66674:20:436)
  at loadPackage (http://192.0.2.1/luci-static/resources/uci.js?v=git-23.292.78392-9f66674:13:38)
  at load (http://192.0.2.1/luci-static/resources/uci.js?v=git-23.292.78392-9f66674:15:76)
  at initNetworkState (http://192.0.2.1/luci-static/resources/network.js?v=git-23.292.78392-9f66674:360:25)
  at getDevices (http://192.0.2.1/luci-static/resources/network.js?v=git-23.292.78392-9f66674:1155:10)
  at load (http://192.0.2.1/luci-static/resources/view/network/routes.js?v=git-23.292.78392-9f66674:1:260)
  at __init__ (http://192.0.2.1/luci-static/resources/luci.js?v=git-23.292.78392-9f66674:147:224)
  at super (http://192.0.2.1/luci-static/resources/luci.js?v=git-23.292.78392-9f66674:22:42)
  at ClassConstructor (http://192.0.2.1/luci-static/resources/luci.js?v=git-23.292.78392-9f66674:12:11)
  at require/compileClass/< (http://192.0.2.1/luci-static/resources/luci.js?v=git-23.292.78392-9f66674:182:111)
  at promise callback*compileClass (http://192.0.2.1/luci-static/resources/luci.js?v=git-23.292.78392-9f66674:178:29)
  at promise callback*require (http://192.0.2.1/luci-static/resources/luci.js?v=git-23.292.78392-9f66674:184:112)
  at instantiateView (http://192.0.2.1/luci-static/resources/ui.js?v=git-23.292.78392-9f66674:358:252)
  at  (http://192.0.2.1/cgi-bin/luci/admin/network/routes:54:7)
  at promise callback* (http://192.0.2.1/cgi-bin/luci/admin/network/routes:53:19)
    raise http://192.0.2.1/luci-static/resources/luci.js?v=git-23.292.78392-9f66674:155
    handleCallReply http://192.0.2.1/luci-static/resources/rpc.js?v=git-23.292.78392-9f66674:15
    promise callback*parseCallReply http://192.0.2.1/luci-static/resources/rpc.js?v=git-23.292.78392-9f66674:9
    promise callback*call http://192.0.2.1/luci-static/resources/rpc.js?v=git-23.292.78392-9f66674:5
    declare http://192.0.2.1/luci-static/resources/rpc.js?v=git-23.292.78392-9f66674:23
    declare http://192.0.2.1/luci-static/resources/rpc.js?v=git-23.292.78392-9f66674:20
    loadPackage http://192.0.2.1/luci-static/resources/uci.js?v=git-23.292.78392-9f66674:13
    load http://192.0.2.1/luci-static/resources/uci.js?v=git-23.292.78392-9f66674:15
    initNetworkState http://192.0.2.1/luci-static/resources/network.js?v=git-23.292.78392-9f66674:360
    getDevices http://192.0.2.1/luci-static/resources/network.js?v=git-23.292.78392-9f66674:1155
    load http://192.0.2.1/luci-static/resources/view/network/routes.js?v=git-23.292.78392-9f66674:1
    __init__ http://192.0.2.1/luci-static/resources/luci.js?v=git-23.292.78392-9f66674:147
    super http://192.0.2.1/luci-static/resources/luci.js?v=git-23.292.78392-9f66674:22
    ClassConstructor http://192.0.2.1/luci-static/resources/luci.js?v=git-23.292.78392-9f66674:12
    compileClass http://192.0.2.1/luci-static/resources/luci.js?v=git-23.292.78392-9f66674:182
    promise callback*compileClass http://192.0.2.1/luci-static/resources/luci.js?v=git-23.292.78392-9f66674:178
    promise callback*require http://192.0.2.1/luci-static/resources/luci.js?v=git-23.292.78392-9f66674:184
    instantiateView http://192.0.2.1/luci-static/resources/ui.js?v=git-23.292.78392-9f66674:358
    <anonymous> http://192.0.2.1/cgi-bin/luci/admin/network/routes:54
    promise callback* http://192.0.2.1/cgi-bin/luci/admin/network/routes:53
luci.js:161:9

```

jow-, 2024-02-20:
Yes, these calls ensure that the relevant uci files are loaded, otherwise all `uci.get()` calls within network.js might yield undefined values.

systemcrash, 2024-02-20:
Understood. Now how do we prevent these noisy tracebacks?