openwrt/luci #6937: acme: config doesn't match what /etc/init.d/acme expects

programmerjake, 2024-02-22:
the GUI sets the `validation_method` option, where the init script expects either the `dns` or `standalone` options instead.

## Steps to reproduce:
**Example: Replace the following lines and remove this**

1. edit /usr/lib/acme/client/acme.sh to just dump the command line arguments and exit, by adding the following lines:
    ```sh
    echo "$0" "$@" >&2
    exit
    ```
2. Create an acme certificate config with the mode set to standalone
3. Save and apply
4. run /etc/init.d/acme start

## Actual behavior:

output:
```
acme-acmesh: Running ACME for example.com
acme-acmesh: /usr/lib/acme/client/acme.sh -d example.com --keylength 2048 --accountemail programmerjake@gmail.com --server letsencrypt --webroot /var/run/acme/challenge --issue --home /etc/acme
/usr/lib/acme/client/acme.sh -d example.com --keylength 2048 --accountemail programmerjake@gmail.com --server letsencrypt --webroot /var/run/acme/challenge --issue --home /etc/acme --pre-hook /usr/lib/acme/notify prepare --renew-hook 
<snip>
```

`/etc/config/acme`:
```
config acme
        option account_email 'programmerjake@gmail.com'
        option debug '0'

config cert 'example'
        option enabled '1'
        option use_staging '0'
        option keylength '2048'
        list domains 'example.com'
        option update_uhttpd '0'
        option validation_method 'standalone'
```

## Expected behavior:

```
acme-acmesh: Running ACME for example.com
acme-acmesh: /usr/lib/acme/client/acme.sh -d example.com --keylength 2048 --accountemail programmerjake@gmail.com --server letsencrypt --standalone --listen-v6 --issue --home /etc/acme
/usr/lib/acme/client/acme.sh -d example.com --keylength 2048 --accountemail programmerjake@gmail.com --server letsencrypt --standalone --listen-v6 --issue --home /etc/acme --pre-hook /usr/lib/acme/notify prepare --renew-hook
<snip>
```

`/etc/config/acme`:
```
config acme
        option account_email 'programmerjake@gmail.com'
        option debug '0'

config cert 'example'
        option enabled '1'
        option use_staging '0'
        option keylength '2048'
        list domains 'example.com'
        option update_uhttpd '0'
        option standalone '1'
```
## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.2'
DISTRIB_REVISION='r23630-842932a63d'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 23.05.2 r23630-842932a63d'
DISTRIB_TAINTS=''
```

stokito, 2024-02-27:
The `validation_method` was added by @jannispinter in 028873054ba3b3d503901e26631ddaa0a1154367
This is a pseudo option used only on UI to show or hide fields and in fact it must not be stored to the UCI config. Instead it must set the `standalone` option but it looks like it wasn't really supported before by the app.

So as a quick fix I can make the form to store set the option `standalone=1` if the `validation_method` method is set to `standalone`.

But I rather not to do that. Anyway you better to use the `webroot` challenge. In fact the `standalone` challenge means only that instead of `uhttpd` webserver the acmesh will use `nc` which is brings as a `netcat` dependency (quite big). This may be useful on some desktop Linux  distro like Ubuntu but makes no sense for OpenWrt. And even in Ubuntu we already have the `busybox httpd` and the netcat is not really needed.

Instead I would propose to change the acme integration and add the `validation_method` option.
Currently the acme.init just loads config, export options as env vars and calls the `hook.sh`.
Inside of the hook [there is a case switch](https://github.com/openwrt/packages/blob/master/net/acme-acmesh/files/hook.sh#L110) that really selects a validation method to use:

```sh
	if [ "$dns" ]; then
		set -- "$@" --dns "$dns"
	elif [ "$standalone" = 1 ]; then
		set -- "$@" --standalone --listen-v6
	else
		set -- "$@" --webroot "$CHALLENGE_DIR"
	fi
```

So if the DNS provider was selected then it will use the DNS validation, then it will check if the standalone mode enabled and only then it will fall back to the webroot validation.

This means that I used DNS validation and then switched to webroot then I have to clean the DNS provider field otherwise it will again use the DNS challenge.
Not a big deal and should really be a problem IRL but still it would be better to just make the switch over the `validation_method` instead. This will allow us to add new challenges easily (like TLS-ALPN-01).

Anyway, not a critical thing as for me. For now we can just remove the standalone option to avoid confusion. If anyone need it and knows what doing then edit config manually.

@tohojo @hgl please have a look.

programmerjake, 2024-02-27:
Ok, I'm using the standalone option to improve security by forwarding the public port 80 to internal port 8080 and editing `hook` to run on port 8080 so the web server is only available when validating, the luci interface is never publicly available which helps defeat attacks through the web server. that kind of config would be really nice to support through the UI...

stokito, 2024-02-27:
Yes, this is a valid case. But if you don't have a publicly available http then maybe you'll use DNS API? There is a big problem that most of DNS providers doesn't have a good and stable API and there is also no any commonly standard. But if you have duckdns or something simple like that then it will be easier and more secure.

systemcrash, 2024-02-27:
@tohojo 

Any ideas? This seems like a problem in the packages repo. The acme-acme.sh hook script even includes this:
```sh

	if [ "$dns" ]; then
		set -- "$@" --dns "$dns"
		if [ "$dalias" ]; then
			set -- "$@" --domain-alias "$dalias"
			if [ "$calias" ]; then
				log err "Both domain and challenge aliases are defined. Ignoring the challenge alias."
			fi
		elif [ "$calias" ]; then
			set -- "$@" --challenge-alias "$calias"
		fi
		if [ "$dns_wait" ]; then
			set -- "$@" --dnssleep "$dns_wait"
		fi
	elif [ "$standalone" = 1 ]; then
		set -- "$@" --standalone --listen-v6
	else
		mkdir -p "$WEBROOT"
		set -- "$@" --webroot "$WEBROOT"
	fi
```

tohojo, 2024-02-28:
I am not opposed to supporting an explicit `validation_method` config variable; however, we need to do so in a backwards-compatible way, so the script needs to handle the case where `validation_method` is *not* set in the config, and fall back to the existing logic. Dunno how hairy that will turn out, but I think it's probably manageable...

stokito, 2024-02-28:
Ok, will send a PR in a few hours

stokito, 2024-02-28:
I created the PR, please review and test