openwrt/luci #6938: luci-app-nlbwmon: Raw values pushed up to human readable ones, seeming way too large

PeterNerlich, 2024-02-22:
## Steps to reproduce:

1. go to: Services → Bandwidth Monitor → Display tab → Traffic Distribution tab

## Actual behavior:

1. Connections, Download (Bytes), Download (Packets), Upload (Bytes) and Upload (Packets) look weird:
  [image]
  because each table cell contains two text nodes that are rendered indistinguishable from one single text node containing both strings, without separation with a space. The numbers thus appear as one, very large number.
  *(screenshot on Chromium Version 122.0.6261.39 (Official Build) Arch Linux (64-bit))*

## Expected behavior:

```
Raw numbers only displayed on mouse hover or otherwise distinguished from the human readable one.
```

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='22.03.2'
DISTRIB_REVISION='r19803-9a599fee93'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 22.03.2 r19803-9a599fee93'
DISTRIB_TAINTS=''
```

luci-app-nlbwmon version: `git-23.119.78260-8aaf581`

jow-, 2024-02-22:
Looks like https://github.com/openwrt/luci/commit/90a2b1eaeb58b7169b31fdc097c5bfe6f557c778 is missing. Did you try clearing your browser cache?  Also a duplicate of #6388

closed 2024-02-22 by jow-

PeterNerlich, 2024-02-22:
Yes thanks, I just saw on another installation that it looked like in your screenshot. I will update luci then.

RoganDawes, 2024-03-15:
Thanks! Was wondering about that (also on 22.03.2). Appreciate the fix, upgraded luci-base and all looking good!