openwrt/luci #6939: Luci wont let me set DNS6 static entry in "Use custom DNS servers" string

Mile-Lile, 2024-02-23:
## Steps to reproduce:
**Example: Replace the following lines and remove this**

1. go to: Network → Interfaces → **wan6** (an interface with a static ipv6 address) → Advanced settings tab 
2. uncheck "Use DNS servers advertised by peer"
3. under "Use custom DNS servers" enter valid DNS6 adress
4. cannot click on green "+" sign and add DNS6 adress - field is red and complains that enry is not valid

## Expected behavior:
LUCI should let me enter DNS6 and should create line in dnsmasq.conf:
**server=2a07:a8c0::75369c**

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r25279-62acd9a2f9'
DISTRIB_TARGET='bcm53xx/generic'
DISTRIB_ARCH='armv7l'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r25279-62acd9a2f9'
DISTRIB_TAINTS='Netgear R8000 (BCM4709)'
```

[image]

jow-, 2024-02-23:
Works for me:
[image]
However I think you simply have a leading space in there, note the different initial space:
[image]

closed 2024-02-23 by jow-

Mile-Lile, 2024-02-23:
ah... ok thx