openwrt/luci #6946: 点击网络报错Runtime error 版本 25.0.3

GuyCui, 2024-03-01:
Runtime error
Unhandled exception during request dispatching
/usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/model/cbi/dockerman/networks.lua:44: attempt to index field 'Config' (a nil value)

In error(), file [C]
called from function [anonymous function] (/usr/lib/lua/luci/ucodebridge.lua:23)
called from function ((tail call))
In [anonymous function](), file /usr/share/ucode/luci/runtime.uc, line 148, byte 45:
  called from function [arrow function] (/usr/share/ucode/luci/dispatcher.uc:794:4)
  called from function render ([C])
  called from function render_action (/usr/share/ucode/luci/dispatcher.uc:768:24)
  called from function run_action (/usr/share/ucode/luci/dispatcher.uc:795:4)
  called from function [anonymous function] (/usr/share/ucode/luci/dispatcher.uc:1008:48)
  called from anonymous function (/www/cgi-bin/luci:39:13)

 `        return lcall.call(modname, method, ...args);`
  Near here ----------------------------------------^

dc-me, 2024-03-06:
I have the same issue, checked the source data[index]["_subnet"] = v.IPAM and v.IPAM.Config[1] and v.IPAM.Config[1].Subnet or nil, because docker has some default networks that do not have subnet and gateway, that's causing the error.

liyk123, 2024-04-02:
I hava the same issue.

dannil, 2024-04-04:
This should be fixed by https://github.com/openwrt/luci/pull/6998, verify on latest `master`.

closed 2024-04-20 by jow-