openwrt/luci #6958: luci-app-firewall: DNAT Port forwards display "IPv4 and IPv6" but rule is only IPv4. Please add option for "IPv4 and IPv6" (family option 'any')

joed74, 2024-03-03:
## Steps to reproduce and Actual behavior:

Add a DNAT forward rule with "automatic":

[image]

That's what luci displaying:

[image]

That's what actually is setup in the firewall:

[image]

## Additional behavior

With  `option family 'any'` added manually to /etc/config/firewall i can get the desired behavior (DNAT rule for both IPv4 and IPv6). But now i cannot touch the rule with luci again (e.g. changing other stuff). If i do so, `option family 'any'` disappears..

## Expected behavior:
If a firewall rule is IPv4 only in the firewall then show the user IPv4 and not "IPv4 and IPv6"

## Desired Extension:

Please add a option "IPv4 and IPv6" to the dropdown box:

[image]

which results in `option family 'any'` in /etc/config/firewall

## Additional Information:

OpenWrt 23.05.2 r23630-842932a63d / LuCI openwrt-23.05 branch git-23.306.39416-c86c256

dannil, 2024-03-03:
This looks to be already added with https://github.com/openwrt/luci/commit/8169f482fd0bb6d711e571e01e565bcc3165469d, right? Try flashing SNAPSHOT and see if it appears, it isn't backported to the 23.05 branch.

systemcrash, 2024-03-03:
I think I cherry picked that fix to 23. In any case I don’t see this as a
bug: any does mean that both families are permitted, even if that interface
has only a single family configured. If the router has IPv6 configured
somewhere this will likely happen.

joed74, 2024-03-03:
@dannil 

I installed this SNAPSHOT package: https://downloads.openwrt.org/snapshots/packages/x86_64/luci/luci-app-firewall_git-24.052.52388-42bd2af_all.ipk

Now there is an option "IPv4 and IPv6" and all is working as expected. Sorry for the report!

closed 2024-03-03 by joed74