openwrt/luci #6962: luci-mod-network: Cipher option is not required for WPA3 Enterprise security mode

morse-sophronia, 2024-03-04:
Cipher option is not required for WPA3 Enterprise security mode, as Openwrt only supports CCMP cipher for wpa3.
[image]

## Steps to reproduce:
[image]

## Actual behavior:
These options for Cipher is not required and it can default to CCMP

## Expected behavior:
if encryption is wpa3, default cipher to CCMP

morse-sophronia, 2024-03-04:
https://github.com/openwrt/luci/pull/6963

systemcrash, 2024-03-04:
I agree with this observation, but there may be newer ciphers in the future which could appear there. So the choice field should remain.

brada4, 2024-03-04:
It should be AES and AES192, certainly no TKIP.
https://www.wi-fi.org/system/files/Wi-Fi_Alliance_Technical_Note_TKIP_v1.0.pdf

Neustradamus, 2024-08-22:
I confirm no TKIP for WPA3:
- WPA3 Personal -> Simultaneous Authentication of Equals (SAE) with forward secrecy (FS/PFS) -> GCMP - AES
- WPA3 Enterprise -> 802.1X/EAP -> GCMP - AES

It must be solved.

webysther, 2025-01-28:
> Cipher option is not required for WPA3 Enterprise security mode, as Openwrt only supports CCMP cipher for wpa3. [image]
>
> ## Steps to reproduce:
> [image]
>
> ## Actual behavior:
> These options for Cipher [...]

Note: The table is from here https://openwrt.org/docs/guide-user/network/wifi/basic#encryption_modes