openwrt/luci #6966: Useless check in 40_luci-nut

oldium, 2024-03-04:
I think the following check never worked actually as intended:

https://github.com/openwrt/luci/blob/48e96d89557cbadfa03ffa351589b6d2273f1ed5/applications/luci-app-nut/root/etc/uci-defaults/40_luci-nut#L3-L6

If the `version` variable has value `2.7.4`, then `release` has to be empty, because it parses the value from the `version` variable, which has no dash `-` inside.

The file `luci/applications/luci-app-nut/root/etc/uci-defaults/40_luci-nut` could be probably completely removed.

systemcrash, 2024-03-04:
It could possibly be removed from master, but is still at least required as long as v22 uses v2.7.4.

The release is derived from the string that opkg -> grep returns (a concat of version '2.7.4' and release e.g. '8' yielding '2.7.4-8').

oldium, 2024-03-04:
@systemcrash I doubt this works. Without actually running the code, I think that the IF always fails. Let's do a mind experiment:

If `version` is `"2.7.4"`, then the following line assigns empty value to release:

```
release="$(echo $version|cut -f2 -d-)"
```

If `version` is `"2.7.4-6"`, then the release is `"6"`.

So the code can lead to just two combinations: `version="2.7.4" release=""` and `version="2.7.4-6" release="6"`. None of this will match the IF statement.

oldium, 2024-03-04:
My conclusion is that this never worked, so either this needs to be fixed, or can be safely removed, because it does not work for years.

oldium, 2024-03-11:
@systemcrash ping

oldium, 2024-03-26:
@systemcrash ping

oldium, 2024-03-29:
@systemcrash

I modified the version in my test to be "2.8.1" because I already have version "2.8.1".

```console
root@h-wrkrm-rtr-pri:~# version="$(opkg status nut|grep 'Version: '|cut -f2 -d\ )"
root@h-wrkrm-rtr-pri:~# echo $version
2.8.1-1
root@h-wrkrm-rtr-pri:~# release="$(echo $version|cut -f2 -d-)"
root@h-wrkrm-rtr-pri:~# echo $release
1
root@h-wrkrm-rtr-pri:~# if [ "$version" = "2.8.1" ] && [ "$release" -lt 8 ]; then echo "Yes"; fi
root@h-wrkrm-rtr-pri:~#
```

Well, it does not work. It _never_ worked.

Do you need any more information from me?

oldium, 2024-04-05:
@systemcrash ping

systemcrash, 2024-04-07:
You just demonstrated that it worked. Perfectly.

hnyman, 2024-04-07:
> demonstrated that it worked. Perfectly.

hmmm.
It doesn't work as intended, I think.
It should have echoed 1 if it had worked as intended.  (main version is 2.8.1 and release 1 is below 8)

Original "version" 2.8.1-1 should be cut to "mainversion" 2.8.1 and "release" 1, like and then mainversion compared to 2.8.1 and release to 8.
`mainversion="$(echo $version|cut -f1 -d-)"`  or something similar.

Now the first part of the example if clause compares full "2.8.1-1" to "2.8.1."

(I happen to know that the original author of the code has submitted several pieces of unworking code in older times, so it is quite possible that the code has been broken since 2018.)

systemcrash, 2024-04-07:
Ok. Jesus. I see it now 🤦🏻‍♂️ mail client line wrap. Well, we've
determined that this code (which perhaps doesn't work as intended) can go
after 22. So I suppose we can bin it from main already.

hnyman, 2024-04-07:
The nut 2.7.4 RELEASE item was bumped to 9 already in 2018.
https://github.com/openwrt/packages/commit/3aa67b3ca3199d7596a4f55134e7389ba7e49184

After that we have had at least nut 2.7.4-9 or greater.
So, 19.07, 21.02, 22.03 and 23.05 users should all have a newer version than the one being compared here.

systemcrash, 2024-04-07:
OK - we're not CPing anything to those branches. 
Working code could have been:
```sh
version="$(opkg status nut|grep 'Version: '|cut -f2 -d\ |cut -f1 -d-)"
release="$(echo "$version"|cut -f2 -d-)"
```
Shall we just bin this script?

hnyman, 2024-04-07:
I removed the script from master and 23.05 (but didn't bother with the old 22.03).

commit 983047a5 (2024-04-07): luci-app-nut: Remove old no-op uci-defaults script

Remove old uci-defaults script that was supposed to be run
if nut was older than 2.4.7-8. Nut has been at 2.7.4-9 or newer
since 2018.

(Ps. The script had faulty logic: the version check never matched.)

Fixes #6966
commit aeb37b43 (2024-04-07): luci-app-nut: Remove old no-op uci-defaults script

Remove old uci-defaults script that was supposed to be run
if nut was older than 2.4.7-8. Nut has been at 2.7.4-9 or newer
since 2018.

(Ps. The script had faulty logic: the version check never matched.)

Fixes #6966

(cherry picked from commit 983047a505d57d72b52e7a291f9a36b4e8c41794)
closed 2024-04-07 by hnyman