openwrt/luci #6973: luci-app-dockerman: networks page error

leobsky, 2024-03-08:
After docker update from 24.0.7 to 25.0.3  the networks page on luci-app-dockerman de error.
https://github.com/openwrt/packages/pull/23482/commits/ef177bae7468197a45f0dd7bc9a0c10f72066002
## Steps to reproduce:

luci-app-dockerman --> networks

## Actual behavior:

[image]

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r25409-4adba64156'
DISTRIB_TARGET='bcm27xx/bcm2712'
DISTRIB_ARCH='aarch64_cortex-a76'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r25409-4adba64156'
DISTRIB_TAINTS=''
```

Joee-D, 2024-03-16:
same problem.
https://github.com/openwrt/luci/blob/72173cce92251aa567b0cbbfa769b944778cd3b7/applications/luci-app-dockerman/luasrc/model/cbi/dockerman/networks.lua#L41
it seems that it uses IPAM config value, but there is no IPAM config, so throw a nil value error.
It should check whether it has the IPAM config first.

closed 2024-03-18 by jow-

systemcrash, 2024-03-18:
Ah, the good old array assumption breakage.

PR #6998 (2024-03-18): luci-app-dockerman: check for existing IPAM config
Fixes #6973 

This was added in upstream with https://github.com/lisaac/luci-app-dockerman/commit/7292955a1b415bb60fa2e403bb3a437b4b7f7846 so should be fine for us to add it.

Networks tab works again after fix. I'm not entirely sure how to test this properly as I don't use Docker via LuCI myself so any test would be appreciated.

[image]

Ping @leobsky @Joee-D

- [X] This PR is not from