openwrt/luci #6979: dnsmasq: Add an option to remove active leases within the gui

Managor, 2024-03-12:
## Steps to reproduce:

1. Go to Network -> DHCP and DNS -> Static Leases
2. Observe the Active DHCP leases

## Actual behavior:
You can only observe them

## Expected behavior:
It should be possible to remove active leases from the GUI

## Additional Information:

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.2'
DISTRIB_REVISION='r23630-842932a63d'
DISTRIB_TARGET='ath79/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='OpenWrt 23.05.2 r23630-842932a63d'
DISTRIB_TAINTS=''

```

dannil, 2024-03-12:
There's some unknowns here:
* Does dnsmasq support terminating an active lease it does it need to be done through editing of the '/tmp/dhcp.leases'?
* Will this actually do anything in practice? Even if you terminate a lease on the server, the clients won't get notified about this, and the next time their lease expires, they'll just request a new one, which will be granted by dnsmasq. Terminating the lease does not equal to blocking the IP address that was leased, or is it that behavior you expect?

Managor, 2024-03-12:
It is the behavior I desire. I'm fully aware that I need to renew the IP on the clients end as well. I just need an easy way to do this within the luci UI.

dannil, 2024-03-12:
Then I understand. I'm unsure if dnsmasq is the proper place to do it since, as I mentioned, terminating a lease won't terminate a client's internet connection, but someone more knowledgeable can pipe in.

systemcrash, 2024-03-12:
Unless dnsmasq itself (which luci relies on) has a trivial way to do this, luci won't ever get one. 

What, by the way, should be the result of 'removing an active lease'? Should it come back? Should it stay removed forever?

JimMatthew, 2024-03-12:
According to [this](https://wiki.gentoo.org/wiki/Dnsmasq#:~:text=Resetting%20leases,-Clients%20that%20had&text=The%20dnsmasq%20service%20stores%20its,and%20start%20the%20service%20again).

To reset a lease early the the lease needs to be removed from the lease file and Dnsmasq needs to be restarted. If that is the case, I don't know if it would be feasible to have the service restart

Managor, 2024-03-12:
> Unless dnsmasq itself (which luci relies on) has a trivial way to do this, luci won't ever get one.
> 
> What, by the way, should be the result of 'removing an active lease'? Should it come back? Should it stay removed forever?

Come back. Often after connecting a device to the network, I reassign it with a static IP. It would be nice to click delete on an active lease within luci so that when I manually set the client to fetch a new IP, it would get an IP based on that static lease.

JimMatthew, 2024-03-12:
> Come back. Often after connecting a device to the network, I reassign it with a static IP. It would be nice to click delete on an active lease within luci so that when I manually set the client to [...]

From my understanding, In this case it sounds like its the client you want to renew the lease. If you assign a new static IP in openwrt, the client won't care until its lease expires, so you would need the client to renew its lease after you change the IP in openwrt.

Managor, 2024-03-12:
It's both. Sure I can do `sudo dhcpcd -k` on a Linux client I have, but not every client has a graceful IP release option like that. A lot of things like printers are very limited in what they can do so disconnecting them from the network, clearing their lease in Openwrt and the reconnecting them to the network so that they fetch a new IP is the only option.

systemcrash, 2024-03-27:
Closing as not possible. Until dnsmasq provides this (for IPv4), we cannot.

closed 2024-03-27 by systemcrash

Managor, 2024-04-12:
Would it not be possible to add a button that simply edits that entry out of `/tmp/dhcp.leases`

carefulcomputer, 2025-01-26:
Please just provide the button to remove the entry out of dhcp lease table. Clients can manage their release/renew on their own (by restart of network stack/device)