openwrt/luci #6983: luci-mod-network: Advanced device options of bridge devices cannot be saved, front end logic locks up after one attempt

qazwsxedcgit, 2024-03-12:
## Steps to reproduce:
1. go to: Network → Interfaces → Devices tab → click "configure" for a bridge device
2. Within the "Bridge device: (device name)" dialog, click "Advanced device options" tab  
3. set "Honor gratuitous ARP" to "enabled"
4. Click the green "Save" button in the bottom right corner of the dialog

## Actual behavior:
"Bridge device: (device name)" dialog stays focused.
No change is saved, i.e. /etc/config/network is not changed at all.

## Expected behavior:
"Bridge device: (device name)" dialog vanishes
Change is saved in the appropriate bridge device section in /etc/config/network

## Additional Information:
After doing this once, no other changes can be made and saved to any other bridge devices on any tabs. Attempts to do so fail as described under "Actual behavior" above. After logging out of LuCI and logging back in, changes are again possible and can be saved again, except on the ""Advanced device options" tab of bridge devices.

DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE=23.05.2'
DISTRIB_REVISION='r23630-842932a63d'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='MediaTek MT7621 ver:1 eco:4'
DISTRIB_DESCRIPTION='LuCI openwrt-23.05 branch (git-23.306.39416-c86c256) / OpenWrt 23.05.2 (r23630-842932a63d)'
DISTRIB_TAINTS=''

JimMatthew, 2024-03-12:
Seems like this is happening on master as well, I am unable to create a new Bridge device with "Honor gratuitous ARP" enabled.

closed 2024-03-12 by systemcrash

JimMatthew, 2024-03-12:
yeah in network.js there is a space after 'arp_accept ' which is causing the issue
o = this.replaceOption(s, 'devadvanced', cbiFlagTristate, 'arp_accept '

PR #6984 (2024-03-12): luci-mod-network: drop trailing whitespace
This resulted in the sysfs parameter to be invalid as it contained a space.

Fixes #6983.