openwrt/luci #7005: luci-base: status overview shows only 2 port staus boxes on Edgerouter Lite (UBNT_E100)

langsdorff, 2024-03-21:
## Steps to reproduce:
Always when opening the status overview page.

## Actual behavior:
The status overview page should show 3 boxes for port status and not only 2.

## Expected behavior:
Other devices I use have the right number of port status boxes.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.2'
DISTRIB_REVISION='r23630-842932a63d'
DISTRIB_TARGET='octeon/generic'
DISTRIB_ARCH='mips64_octeonplus'
DISTRIB_DESCRIPTION='OpenWrt 23.05.2 r23630-842932a63d'
DISTRIB_TAINTS=''
```

Partial output of `ip addr`

1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
...
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-lan state UP group default qlen 1000
...
3: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
...
4: eth2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel master br-lan state UP group default qlen 1000
...

jow-, 2024-03-21:
Please provide the output of `ubus call luci getBuiltinEthernetPorts` and `cat /etc/board.json`

langsdorff, 2024-03-21:
`ubus call luci getBuiltinEthernetPorts`
{
        "result": [
                {
                        "role": "lan",
                        "device": "eth0"
                },
                {
                        "role": "wan",
                        "device": "eth1"
                }
        ]
}

`cat /etc/board.json`
{
        "model": {
                "id": "erlite",
                "name": "UBNT_E100 (CN5020p1.1-500-SCP)"
        },
        "network": {
                "lan": {
                        "device": "eth0",
                        "protocol": "static"
                },
                "wan": {
                        "device": "eth1",
                        "protocol": "dhcp"
                }
        }
}

jow-, 2024-03-21:
This is not a LuCI issue then. The OpenWrt generated board.json should include all ethernet ports.