openwrt/luci #7007: Preparations to switch to APK package manager

aparcar, 2024-03-21:
Hi all, some fellow developers and me worked for some time now on making [APK](https://gitlab.alpinelinux.org/alpine/apk-tools) the new package manager for OpenWrt, replacing the unmaintained OPKG fork we've been using for the longest time.

APK is actively developed and used in multiple other distributions, e.g. Alpine Linux 🎉 

While there is till some work ahead, I'd like to prepare everyone who maintains a package to verify that the `PKG_VERSION` follows [Semantic Versioning](https://semver.org). APK uses a deterministic algorithm to compare versions and does not like random strings, except a valid hash prefixed with a `~`.

If in doubt, please use the Docker container below to verify the used version is valid:

```
docker run --rm -it docker.io/aparcar/apk-version-check <VERSION> [<VERSION> ...]
```

It will print whatever version is not valid, if you get a zero exit code, you're fine.

Please feel free to reach out for assistance and have a look at the [core migration of versions](https://github.com/openwrt/openwrt/pull/14918).

aparcar, 2024-03-21:
Below packages that would cause trouble. Please bear in mind that PKG_RELEASE is soonish prefixed with an r and may only contain a number.

**Naughty list:**

- [ ] luci-app-adblock-fast            `1.1.1-5`
- [ ] luci-app-advanced-reboot         `1.0.1-9`
- [ ] luci-app-dockerman               `v0.5.13-20240317`
- [ ] luci-app-https-dns-proxy         `2023-11-19-1`
- [ ] luci-app-pbr                     `1.1.1-7`
- [ ] luci-proto-nebula                `1.6.1-1`

stangri, 2024-03-21:
@aparcar I can pre-emptively add r to the revisions, I just wanted to confirm that it will not break opkg upgrade logic. For example, would opkg consider 1.1.1-r6 higher version than 1.1.1-5?

hnyman, 2024-03-21:
> would opkg consider 1.1.1-r6 higher version than 1.1.1-5? Same question for 2023.11.19-r2 and 2023-11-19-1.

Easiest is to bump the main version to 1.1.2 at the same time

(not sure about the date comparison with a changed separator)

systemcrash, 2024-03-21:
> @aparcar I can pre-emptively add r to the revisions, I just wanted to confirm that it will not break opkg upgrade logic. For example, would opkg consider 1.1.1-r6 higher version than 1.1.1-5? Same question for 2023.11.19-r2 and 2023-11-19-1.

Does opkg even do that? I'm under the impression that it does not compare versions. 

See [parse_version](https://git.openwrt.org/?p=project/opkg-lede.git;a=blob;f=libopkg/pkg_parse.c;h=c9e0cdf45a4236afad622041877bd1578f985241;hb=d038e5b6d155784575f62a66a8bb7e874173e92e#l64) and [compare](https://git.openwrt.org/?p=project/opkg-lede.git;a=blob;f=libopkg/opkg_cmd.c;hb=d038e5b6d155784575f62a66a8bb7e874173e92e#l1312)

You could do away with 1.1.1-r6 altogether and just let git handle semantic versioning (at least for the time being), or use dates instead. For devs, the tip of the master branch is latest, for users, if it is updated, it will show as update available.

systemcrash, 2024-03-21:
@aparcar 

@feckert: date format here makes most sense. We won't be updating lua apps (this app is lua) any more so this change should be OK. Alt: update at source? https://github.com/lisaac/luci-app-dockerman

- [ ] luci-app-dockerman               `v0.5.13-20240317`

stangri, 2024-03-21:
> docker run --rm -it docker.io/aparcar/apk-version-check <VERSION> [<VERSION> ...]

BTW:

```sh
sudo docker run --rm -it docker.io/aparcar/apk-version-check 1.1.1-5
Unable to find image 'aparcar/apk-version-check:latest' locally
latest: Pulling from aparcar/apk-version-check
bca4290a9639: Pull complete
1ae19ff60baf: Pull complete
75e05e2c02ea: Pull complete
28c83907721f: Pull complete
Digest: sha256:e113fde83c649c3dc7cfed9b7e6c2068b94aad61f950cf9f0bfd93d9bcf69c86
Status: Downloaded newer image for aparcar/apk-version-check:latest
WARNING: The requested image's platform (linux/arm64/v8) does not match the detected host platform (linux/amd64/v2) and no specific platform was requested
exec /usr/bin/apk: exec format error
```

aparcar, 2024-03-21:
@stangri I added a multi arch container here `ghcr.io/aparcar/apk-valid-version`

jow-, 2024-03-21:
> For example, would opkg consider 1.1.1-r6 higher version than 1.1.1-5? Same question for 2023.11.19-r2 and 2023-11-19-1.

```
root@OpenWrt:~# opkg compare-versions 1.1.1-r6 '>' 1.1.1-5 && echo true || echo false
true
root@OpenWrt:~# opkg compare-versions 2023.11.19-r2 '>' 2023-11-19-1 && echo true || echo false
true
root@OpenWrt:~# opkg compare-versions 1 '>' r1 && echo true || echo false
false
root@OpenWrt:~# opkg compare-versions r1 '>' 1 && echo true || echo false
true
root@OpenWrt:~# 
```

stangri, 2024-03-23:
Thank you @aparcar @hnyman @jow-!

PRs with fixes:
* luci-app-adblock-fast: https://github.com/openwrt/luci/pull/7009
* luci-app-advanced-reboot: https://github.com/openwrt/luci/pull/7010
* luci-app-https-dns-proxy: https://github.com/openwrt/luci/pull/7013
* luci-app-pbr: https://github.com/openwrt/luci/pull/7011
* luci-proto-nebula: https://github.com/openwrt/luci/pull/7012

systemcrash, 2024-03-27:
@feckert - how is dockerman update coming along?

olumolu, 2024-08-02:
Status of dockerman will this be ready on 24.x

krant, 2024-08-18:
@stangri latest luci-app-advanced-reboot fails to build with `CONFIG_USE_APK=y` 
```
ERROR: info field 'version' has invalid value: package version is invalid
```

hnyman, 2024-09-01:
@aparcar 
I started looking into trying to convert luci-app-opkg to a crude initial luci-app-apk, but stumbled rather quickly into trouble. The apk package info is formatted so differently than opkg, that the LuCI app requires major javascript work to even get it to read the available packages list and install new packages.

One challenge will be that our apk seems to produce rather small amount of information for available packages with "apk list -a" (and for installed ones with "apk list -I")

In style of
```
root@router6000:~# apk list -a
adblock-4.2.2-r3 all {feeds/packages/net/adblock} (GPL-3.0-or-later) [installed]
apk-openssl-3.0.0_pre20240806-r1 aarch64_cortex-a53 {feeds/base/system/apk} (GPL-2.0-only) [installed]
arptables-nft-1.8.8-r2 aarch64_cortex-a53 {feeds/base/network/utils/iptables} (GPL-2.0)
...
wifi-scripts-1.0-r1 all {feeds/base/network/config/wifi-scripts} (GPL-2.0) [installed]
wireguard-tools-1.0.20210914-r3 aarch64_cortex-a53 {feeds/base/network/utils/wireguard-tools} (GPL-2.0) [installed]
wireless-regdb-2024.07.04-r1 all {feeds/base/firmware/wireless-regdb} (ISC) [installed]
wpad-openssl-2024.03.09~695277a5-r3 aarch64_cortex-a53 {feeds/base/network/services/hostapd} (BSD-3-Clause) [installed]
xdp-filter-1.4.2-r1 aarch64_cortex-a53 {feeds/base/network/utils/xdp-tools} (GPL-2.0-only)
xdp-loader-1.4.2-r1 aarch64_cortex-a53 {feeds/base/network/utils/xdp-tools} (GPL-2.0-only)
```

Trouble starts already with package name and version being shown in combined form. I have not yet figured a way to have those listed separately by apk list. We might need to assume some regexp like `^([\w-]+)-(\d[\w\.\-~]*) (.[\w\-]*) (.*) (\(.*\))( \[installed\])*$` to parse the attributes from the single line into fields. Big change compared to opkg where each field was on its own row with a label.

Ps. Is there any support for OpenWrt specific definitions like essential, hold, etc.?  "apk info -a" for an individual package shows more details, but so far I have not seen any OpenWrt specific attributes.

dannil, 2024-10-04:
I haven't seen any news about it but do we know if 24.x will default to using APK? In that case we still need to fix `luci-app-dockerman`'s version string, it's quite a popular app as well, shows up time to time on the form.

olumolu, 2024-10-05:
> I haven't seen any news about it but do we know if 24.x will default to using APK? In that case we still need to fix `luci-app-dockerman`'s version string, it's quite a popular app as well, shows up time to time on the form.

This seems to be unmaintained for more than a year.

systemcrash, 2024-11-12:
So how about `0.5.13.20240317`? That seems like it would work, but the current installs may not recognise this 'version' jump, and might require a forced reinstall. Acceptable compromise I think.

AzimsTech, 2024-11-13:
[image]

The `/etc/config/luci` file is broken, it should contain a set of theme entries that are added when a luci-theme-* package is installed.

```
root@OpenWrt:~# cat /etc/config/luci
config core main
        option lang auto
        option mediaurlbase /luci-static/bootstrap
        option resourcebase /luci-static/resources
        option ubuspath /ubus/

config extern flash_keep
        option uci              "/etc/config/"
        option dropbear "/etc/dropbear/"
        option openvpn  "/etc/openvpn/"
        option passwd   "/etc/passwd"
        option opkg             "/etc/opkg.conf"
        option firewall "/etc/firewall.user"
        option uploads  "/lib/uci/upload/"

config internal languages

config internal sauth
        option sessionpath "/tmp/luci-sessions"
        option sessiontime 3600

config internal ccache
        option enable 1

config internal themes

config internal apply
        option rollback 90
        option holdoff 4
        option timeout 5
        option display 1.5
```

stangri, 2024-11-13:
Maybe there's a different reason, but I'm guessing apk sets the `$PKG_UPGRADE` to 1 and/or the entire [uci-defaults script](https://github.com/openwrt/luci/blob/master/themes/luci-theme-material/root/etc/uci-defaults/30_luci-theme-material) fails.

@AzimsTech if you can verify that the theme apk binaries contain the uci-defaults scripts that would be a good start.

AzimsTech, 2024-11-13:
@stangri I get it now. I just need to reboot for the configuration to apply. I was not aware on how it works at first. I'm sorry for the false alarm. 🙏

famewolf, 2024-12-05:
Just wanted to report that on a recent snapshot the apk for statistics has some missing icons.  See attached.
[image]

famewolf, 2024-12-11:
> > statistics has some missing icons
> 
> Nothing to do with apk. That is about mis-detected temperature sensors.
> 
> Just define correctly the sensors in your system that actually do provide temperature info.

I can pick just one zone and those images are still not valid.

hnyman, 2024-12-11:
The icons get created if there is old/incompatible RRD files. 
You may have to remove the old RRD database files, if they do not match the current sensors or do not provide data.  
But again, not related to apk. Just about LuCI statistics being rather simplistic.

systemcrash, 2025-01-05:
@aparcar @Ansuel 

Can I get some input for a version number check? I think to implement a CI check to ensure version numbering doesn't stray from the established norm here. I'm working with the notion that it goes `<major>[.<minor>[.<point>]]` e.g.
```sh
grep -hr '^PKG_VERSION:=' --include="Makefile" | grep -vE '^PKG_VERSION:=[0-9]+(?:\.[0-9]+(?:\.[0-9]+)?)?$'
```

This works within the luci repo. Am I right in thinking that an SHA hash is also possible? I think we can discount it since we don't pull in external repos or downloads for the luci repo, but maybe someone tracks a remote branch. Worst case is that a PR check can be ignored or overridden.

The incantations in the openwrt repo include things like:
```
PKG_VERSION:=c99c8d491850dc3a6e0b8604a2729d8bc5c0eff1# # stable-202401
PKG_VERSION:=$(PKG_VERSION)
PKG_VERSION:=$(LINUX_VERSION)
...
PKG_VERSION:=$(subst _,.,$(subst -,.,$(PKG_VERSION_REAL)))
PKG_VERSION:=6.6.3.1.0.0
...
PKG_VERSION:=05.08.01.08.01.06.05.08.00.11.01.01
...
PKG_VERSION:=1.6.10-openwrt
PKG_VERSION:=2020-03-11
PKG_VERSION:=6.6.23.2.0.0
PKG_VERSION:=v4.0.3
PKG_VERSION:=linux4sam-2022.04
...
```

hnyman, 2025-01-06:
@systemcrash 
I think that there is no need for a separate check, as the package build process already verifies the version. The .apk package creation fails if version does not follow the rules.

(Note that there was a crude CI logic in the packages repo, but that was removed as unnecessary when APK was made default. And it was faulty in any case as it failed with those subst calculations)

If you really want to look into it, please check the apk sources for their interpretation on semantic versioning. It is much more complex logic.

aparcar, 2025-01-06:
@systemcrash the logic is a bit more involved, see [here](https://gitlab.alpinelinux.org/alpine/apk-tools/-/blob/master/doc/apk-package.5.scd?ref_type=heads#L47). What could be done is making use of the `make package/foobar/check V=s` command and let it run `apk` to verify the version itself. The check is right now included as @hnyman said, however it's buried somewhere deep down 30k lines of compilation code.

closed 2026-02-04 by systemcrash