openwrt/luci #7022: luci-app-ttyd: Nginx reverse proxy

alex9434, 2024-03-23:
## Steps to reproduce:

1. Connect to Luci with cloudflared or behind firewall with only 80 and 443 open ports
2. Go to services -> terminal

## Actual behavior:
Empty page for terminal

## Expected behavior:
Terminal windows

Lucy-app-ttyd uses separate port (default 7681) for ttyd. I would like to use nginx reverse proxy to have ttyd on port 80 (and 443). That requires to change the web interface to use a path for ttyd to use nginx reverse proxy.

ysc3839, 2024-03-25:
This issue has been there for some years. Because there are many different SSL/reverse proxy configurations. It's difficult to detect it automatically. See also #5347.

alex9434, 2024-03-25:
Hi @ysc3839 
My suggestion is different than #5347 . I want to use nginx as reverse proxy (could be an option in the config)

```
        location /ttyd {
                proxy_set_header Host $host;
                proxy_set_header X-Forwarded-Proto $scheme;
                proxy_set_header X-Forwarded-Port $server_port;
                proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
                proxy_pass http://localhost:7681;
                proxy_http_version 1.1;
                proxy_set_header Upgrade $http_upgrade;
                proxy_set_header Connection "upgrade";
        }
```
However, that would require the web-interface to use the path /ttyd for requests.

alex9434, 2024-03-27:
Great, will test and report back

alex9434, 2024-04-16:
Hi @ysc3839, I have successfully compiled and installed your commit. Apologies for the delay. I have added the above snipped to `/etc/nginx/conf.d/ttyd.locations` (and restarted nginx).

What exactly do I need to put in `URL_override` in the LuCI config for ttyd? Just putting in `/ttyd` does not seem to work. Other other options that I need to chose to make this work?

ysc3839, 2024-04-16:
Did you start ttyd? The error is `Connection refused`, which means no program is listening that port.
And you should use UNIX socket for localhost reverse proxy, because it has better performance.

alex9434, 2024-04-17:
Hi @ysc3839, ttyd is running but without active instances.
```
service ttyd status
active with no instances
```
Which is strange because I have created an instance that is enabled in LuCI. When I create an instance with Unix socket in `/var/run/ttyd.sock`, the file does not appear. What is going on?

alex9434, 2024-05-09:
Hi @ysc3839 apologies for the late reply. It took some time for me to build OpenWrt again and test the changes. I can confirm that this is now working! For everybody that is interested, here is what I did:
nginx: ttyd.locations
```
location ~ ^/ttyd(.*)$ {
	auth_digest 'Restricted Area'; # set the realm for this location block
	auth_digest_user_file passwd.digest;
	proxy_http_version 1.1;
	proxy_set_header Host $host;
	proxy_set_header X-Forwarded-Proto $scheme;
	proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
	proxy_set_header Upgrade $http_upgrade;
	proxy_set_header Connection "upgrade";
	proxy_pass http://unix:/var/run/ttyd.sock:/$1;
}
```
UCI
```
uci set ttyd.@ttyd[0].interface=' /var/run/ttyd.sock'
uci set ttyd.@ttyd[0].url_override='/ttyd'
uci set ttyd.@ttyd[0].unix_sock='1'
uci commit ttyd
```
It works perfect on https and cloudfared. Thank you very much for the change. I support merging this.

logread, 2025-04-05:
> auth_digest 'Restricted Area'; # set the realm for this location block
> 	auth_digest_user_file passwd.digest;

this makes error in my setup using default openwrt `nginx-full` package, default package doesn't include required module i guess