openwrt/luci #7024: luci-nginx: non `-ssl` version still enables SSL?

ysc3839, 2024-03-25:
## Steps to reproduce:
1. Using latest snapshot image
2. `opkg install luci-nginx`
3. Open `http://<device-ip>` in browser

## Actual behavior:
Redirected to `https://<device-ip>`, and browser warns invalid certificate

## Expected behavior:
Load `http://` page normally.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r25659-004eac14ab'
DISTRIB_TARGET='armsr/armv8'
DISTRIB_ARCH='aarch64_generic'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r25659-004eac14ab'
DISTRIB_TAINTS=''
```
```
# opkg depends nginx-mod-luci
nginx-mod-luci depends on:
	libc
	uwsgi
	uwsgi-luci-support
	nginx-ssl
	nginx-mod-ubus
```
`nginx-mod-luci` depends on `nginx-ssl`. Is this an expected behavior?

dannil, 2024-03-27:
`nginx-mod-luci` is actually part of [openwrt/packages](https://github.com/openwrt/packages/blob/master/net/nginx/Makefile#L182) so I guess you should report the issue there instead? If that's what you're suspecting is configured incorrect.

jow-, 2024-04-20:
Upstream forcibly enables SSL support in nginx, there's no way to provide non-ssl variants of it anymore.

commit 218bdbb8 (2024-04-20): collections: drop luci-ssl-nginx

Since upstream commit
https://github.com/openwrt/packages/commit/f5d7a26531aa0e6fdade9836dfef3450fb007b01
it is not possible to ship non-ssl nginx flavors anymore.

Drop the luci-ssl-nginx collection, retain just luci-nginx and let it provide
a virtual luci-ssl-nginx package for backwards compatibility.

Fixes: #7024
closed 2024-04-20 by jow-