openwrt/luci #7027: luci-theme-material: system log "Scroll to tail" and "Scroll to head" buttons doesn't work

dannil, 2024-03-25:
## Steps to reproduce:

1. Go to: Status -> System Log
2. Click either "Scroll to tail" or "Scroll to head" buttons

## Actual behavior:

Clicking "Scroll to tail" and "Scroll to head" doesn't do anything

## Expected behavior:
Clicking "Scroll to tail" should jump to the bottom of the page and "Scroll to head" should jump to the top of the page, just as the Bootstrap theme does.

## Additional Information:

I also saw this issue in SNAPSHOT before the `23.05.3` release as well so has been there for a while at least.

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.3'
DISTRIB_REVISION='r23809-234f1a2efa'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 23.05.3 r23809-234f1a2efa'
DISTRIB_TAINTS=''
```

systemcrash, 2024-03-26:
confirmed

systemcrash, 2024-03-26:
Actually, this is only a problem when the side-panel is visible. If I open the debug window and widen it enough or narrow the window enough such that the side-panel (automatically) disappears, then the buttons work.

McGiverGim, 2024-03-27:
The problem is that OpenWrt can use themes. The actual code moves the scroll of the "window", that is were the bootstrap theme has the scroll:
https://github.com/openwrt/luci/blob/b69bce077d7d198f9618881b79e1a13b878d466f/modules/luci-mod-status/htdocs/luci-static/resources/view/status/syslog.js#L30-L32

That's true for the bootstrap theme, but not for material or argon theme, this and other themes use DIV for the scroll.

I've a custom code that "searchs" for a scroll in the parents and moves it. It works but I don't know if under some circumstances can move a "wrong" scroll.

I will do more tests and push a PR to let others test it and see if it works or not.

closed 2024-03-27 by systemcrash