openwrt/luci #7030: luci-app-acme: Page is not loading

JKSTAFF, 2024-03-26:
## Steps to reproduce:
**Example: Replace the following lines and remove this**
1. Make a fiemware with luci-app-acme and its depends build in
2. go to: Services → ACME certs

## Actual behavior:
In browser luci shows 
```
RPCError
RPC call to uci/get failed with ubus code 9: Unspecified error
  at ClassConstructor.handleCallReply (https://192.168.0.1/luci-static/resources/rpc.js?v=git-24.080.71710-b07cf9d:15:3)
```

## Expected behavior:
ACME gui displayed correctly

## Additional Information:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05-SNAPSHOT'
DISTRIB_REVISION='r23812-0a2047cf77'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 23.05-SNAPSHOT r23812-0a2047cf77'
DISTRIB_TAINTS='no-all'
```

systemcrash, 2024-03-26:
@tohojo any ideas?

> Make a fiemware with luci-app-acme and its depends build in

@JKSTAFF and what are its depends?

tohojo, 2024-03-26:
No idea how the UCI stuff works under the hood, sorry :/

JKSTAFF, 2024-03-26:
> @tohojo any ideas?
> 
> > Make a fiemware with luci-app-acme and its depends build in
> 
> @JKSTAFF and what are its depends?

I checked that `luci-mod-admin-full` `acme` `acme-acmesh` `acme-common` `acme-acmesh-dnsapi` `libc` were installed

jow-, 2024-03-26:
Error code 9 is `UBUS_STATUS_UNKNOWN_ERROR` - this can only happen if libuci reports a lower level error such as bus error.