openwrt/luci #7035: luci-mod-status: Incomplete Port Status display for BCM57810 with PCI Passthrough on Proxmox VE

missing233, 2024-03-27:
## Steps to reproduce:

Pass through the BCM57810 to OpenWrt.
[image]

## Actual behavior:
When using Proxmox VE's PCI passthrough feature to assign a Broadcom BCM57810 which has two Ethernet ports to an OpenWrt virtual machine, the LuCI interface only shows the port status for one interface. The second interface, although operational and recognized by the system, does not appear in the Port Status.
[image]
[image]
In the `lspci` command, BCM57810 is output normally.
[image]

## Expected behavior:
LuCI should display all ports.

## Additional Information:
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='OpenWrt 23.05.2'
DISTRIB_REVISION='r23630-842932a63d'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 23.05.2 r23630-842932a63d'
DISTRIB_TAINTS='no-all'
```

systemcrash, 2024-03-27:
Broadcom is garbage. Try something else.

dannil, 2024-03-27:
I've never worked with Proxmox, so just a random shot in the dark, but is `eth2` listed under `/sys/class/net`? If it is, does `ls -la /sys/class/net/eth2` resolve to the correct PCI device?

missing233, 2024-03-27:
> I've never worked with Proxmox, so just a random shot in the dark, but is `eth2` listed under `/sys/class/net`? If it is, does `ls -l /sys/class/net/eth2` resolve to the correct PCI device?

Looks fine?
[image]

missing233, 2024-03-27:
> Broadcom is garbage. Try something else.
> 
> Edit: since broadcom has poor opensource support, you might have to emulate instead.

> Broadcom is garbage. Try something else.
> 
> Edit: since broadcom has poor opensource support, you might have to emulate instead.

Looks like only the BCM57810 supports 2.5Gbps SFP.
So sad :(

systemcrash, 2024-03-27:
Hmm, do each of the sockets have an SFP device inserted? The devices tab shows no specific config for it yet.

missing233, 2024-03-27:
> Hmm, do each of the sockets have an SFP device inserted? The devices tab shows no specific config for it yet.

Yes. eth1 is used as the WAN interface connected to the ONU, and eth2 is in br-lan, connected to a switch.

jow-, 2024-03-27:
Well LuCI only displays ports enumerated in /etc/board.json with an additional exception for x86-64 where it also displays all devices starting with "eth" using the same bus as "eth0".

Your passed through devices neither are in /etc/board.json nor do they use the same bus as the already present eth0, so the heuristic does not trigger.

I do not have plans to fix this, you can work around the issue by manually adding the additional eth* devices to /etc/board.json

closed 2024-03-28 by jow-