openwrt/luci #7043: luci-app-siitwizard: web page shows "Model 'clamav' not found"

zwhfly, 2024-04-04:
## Steps to reproduce:

1. `opkg install luci-app-siitwizard`
1. go to: Network → SIIT 4over6 assistant

## Actual behavior:

> Runtime error
> Unhandled exception during request dispatching
> /usr/lib/lua/luci/ucodebridge.lua:23: /usr/lib/lua/luci/cbi.lua:53: Model 'clamav' not found!

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.2'
DISTRIB_REVISION='r23630-842932a63d'
DISTRIB_TARGET='mediatek/filogic'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 23.05.2 r23630-842932a63d'
DISTRIB_TAINTS=''
```

## Probable Cause:
Should the "clamav" be changed to "siitwizard" in this line?

https://github.com/openwrt/luci/blob/835ac7529e5cbd5a855fbc3ae07033b11e8d4ed3/applications/luci-app-siitwizard/root/usr/share/luci/menu.d/luci-app-siitwizard.json#L7

https://github.com/openwrt/luci/commit/625abbfdbe94dada023ebd8ea69778173d1a7351#diff-f44880b78c0e61b0fa3bed9999472416568a145bc4f3a5976385a2a19b6a1ef2R7

systemcrash, 2024-04-04:
It certainly looks like it. I can amend that, and you can see whether that fixes your problem. 

I don't even know what siitwizard does...

closed 2024-04-04 by systemcrash

zwhfly, 2024-04-17:
> It certainly looks like it. I can amend that, and you can see whether that fixes your problem.

I edited the file `luci-app-siitwizard.json` like in your commit, and the web page showed up successfully.

> I don't even know what siitwizard does...

Neither do I. It turned out what I need is kmod-nat46 instead of kmod-siit, so I stopped digging siit side of things.

zwhfly, 2024-04-17:
First of all thanks for your commit.

> It certainly looks like it. I can amend that, and you can see whether that fixes your problem.

I edited the file `luci-app-siitwizard.json` like in your commit, and the web page showed up successfully.

> I don't even know what siitwizard does...

Neither do I. It turned out what I need is kmod-nat46 instead of kmod-siit, so I stopped digging siit side of things.

commit a54fcc59 (2024-04-04): luci-app-siitwizard: fix copy paste error from app conversion

Closes #7043