openwrt/luci #7059: luci-base: during development, static files are often cached

wryun, 2024-04-12:
uhttpd doesn't set a Cache-Control header, which I believe means the browser defaults to heuristics:
https://webmasters.stackexchange.com/questions/111298/what-happens-if-you-dont-set-cache-control-header#comment145064_111298

For actual releases, this is fine, as LuCI includes a cache-busting parameter in the URL. However, during development, it's often useful to copy JS files (and other files) directly to the device. i.e. I think it would be great if we had this patch:

https://patchwork.ozlabs.org/project/openwrt/patch/20180704132659.5880-2-fe@dev.tdt.de/

Is there a workaround for this that I'm missing? At the moment, I'm disabling the cache in the browser (e.g. via devtools), but this means it ignores etags and fetches every time (annoying).

stokito, 2024-04-16:
Check https://github.com/openwrt/luci/wiki/DevelopmentEnvironmentHowTo

>  it is recommended to use `uci set luci.ccache.enable=0; uci commit luci` in order to disable the code caching for development purposes.

wryun, 2024-04-16:
My understanding is that this disables the lua code caching, and doesn't do anything about the assets (including the JS).

remittor, 2024-10-27:
> ...and doesn't do anything about the assets (including the JS).

I had a similar question/problem: https://forum.openwrt.org/t/214025

wryun, 2024-11-13:
@remittor indeed, a similar problem if you're just upgrading an individual package. Might be nice to cache bust based on a hash of all the installed package versions rather than have the hash built into luci-base (what currently happens due to how require works, AFAIK)...

remittor, 2025-04-14:
@wryun 
> Might be nice to cache bust based on a hash of all the installed package versions rather than have the hash built into luci-base. i.e. dynamically feed in thie better value via `header.ut`, and use it if available.

Fix: https://github.com/openwrt/luci/pull/7725

systemcrash, 2025-04-14:
Closed by #7725

wryun, 2025-07-21:
@systemcrash could we re-open this?