openwrt/luci #7064: luci: [RFE] change root redirection from meta refresh to HTTP redirect

Lanchon, 2024-04-15:
i'm a control freak, no doubt. that's why i use openwrt and not stock.

it's also why i forbid my browser from automatically follow redirections. (and many pages refresh while you are reading them just to impress more ads -not that they'd show in my browser- and it's annoying as hell.)

so when accessing the root of any openwrt browser i get this:

[image]

i've clicked through this page hundreds of times and it's... well, not as exciting as it once was.

it's a very tiny usability wrinkle, but i still wanted to ask that the root file at `/` containing the meta refresh to `/cgi-bin/luci/` be replaced if possible with a 302 [HTTP redirect](https://developer.mozilla.org/en-US/docs/Web/HTTP/Redirections).

btw, in the previous link Mozilla has this to say regarding HTML redirect (aka meta refresh):

> HTTP redirects are the best way to create redirections, but sometimes you don't have control over the server. In that case, try a `meta ` element with its `http-equiv` attribute set to `Refresh in the `head` of the page.

thank you!

Lanchon, 2025-01-15:
> browsers permanently remember the redirection because the meta refresh was requested with zero delay. It is annoying to access a different web server with the same IP address in a different network because the browser tries to jump to /cgi-bin/luci/.

no, not really. your browser is just correctly caching the cacheable resource in `/`, because openwrt said it was cacheable. this is a side effect of using IP incorrectly: you should never talk to different computers with the same IP address! for example, your home network should never be at 191.168.(0|1).0. you have huge private IP pools to choose from and avoid clashes. use them!

even if you don't control them... well, they are still messed up.

but you *can* provide local names for the computers you use. ideally they should be on different domains. if you access them by name instead of IP, then your browser will not cross-cache them even if the names map to the same IP.

> It should be elegant to replace [the meta tag](https://github.com/openwrt/luci/blob/ae8bbb814f16c2dc2a82e88d408bee1812ceda9c/modules/luci-base/root/www/index.html#L19) with a single line of JavaScript to do the equivalent redirection with a short delay.

absolutely not. meta refresh can use a delay, but that should not be used here. and it wouldn't help in any way with the caching issue. a JS script to do a redirection is the absolute worse solution and fortunately i'm pretty sure that it will never happen.

your problem (and also mine) can simply be solved by a non-cacheable 302 HTTP redirect.

puhitaku, 2025-01-15:
I'm not saying this without no evidence and what's happening in the real world is what a browser does. And furthermore, I'm not saying that I'm incorrectly using the same IP address in the *same* network. The same IP address can happen in a *different* network. Am I saying anything incorrect?

Anyway, okay, I get chill here... So what WE should achieve here is to do a 302 from `/` to somewhere which is mentioned in the source. Do we have to edit largely to achieve this?

puhitaku, 2025-01-15:
I'm honestly totally new to Lua and still missing a way to contribute a change. I'd be happy if a possible plan or a pointer to what's gonna be edit here is pointed out. Thanks.

stangri, 2025-01-15:
@Lanchon I don't think `uhttpd`, which is a default web-server in OpenWrt supports redirect/rewrite instructions. You may be able to achieve what you want (without adding a refresh exception to your browser) by installing and configuring nginx instead of uhttpd -- have you tried that?

Lanchon, 2025-01-15:
> I don't think `uhttpd`, which is a default web-server in OpenWrt supports redirect/rewrite instructions.

oh that's bad. i didn't know the server didn't implement basic HTTP stuff. well if that's true, then there is no fixing this.

Lanchon, 2025-01-15:
> The same IP address can happen in a _different_ network.

correct. many networks can use the same IPs.

but...

you shouldn't be connecting to such networks! among the network *you* use, there *should* be no IP clashes. if there are, the stuff will break. like your browser.

so no, you don't need to fix openwrt, openwrt is fine. you need to fix your networks.