openwrt/luci #7089: i18n updates

systemcrash, 2024-04-27:
Proposal
FAO: @hnyman @aparcar @stokito @stangri 

To prevent/remove location `#: applications/luci-app-blah/...:374` lines, via `msgmerge --no-location`.

In our automated systems, these lines are responsible for change and churn. Does the JS GUI need them?
This can be applied to templates and po files.

A second optimisation is altogether dropping the pruned lines `#~ ...`.

hnyman, 2024-04-27:
I asked about removing those location lines from .po files a few years ago, but if I remember right, @jow- thinks that they are necessary.

hnyman, 2024-04-27:
> second optimisation is altogether dropping the pruned lines #~ ...

I have actually done that manually a few times for the .po files where has been lots of pruned lines.

hnyman, 2024-04-27:
Reference to earlier issue about this.

https://github.com/openwrt/luci/issues/5655

systemcrash, 2024-04-27:
I mean, does anyone manually translate using poedit? Or does everything go via weblate now?

systemcrash, 2024-04-27:
I think a better solution for any outliers using poedit is ... another script :D 
A script which pre-processes what they want, adds the locations, job done, remove locations. It's a workflow admittedly, but, I dunno. What do others think?

systemcrash, 2024-04-27:
> > second optimisation is altogether dropping the pruned lines #~ ...
> 
> I have actually done that manually a few times for the .po files where has been lots of pruned lines.

We can at least fix this in the perl scripts to drop those lines. The commit history shows the changes anyway.

stokito, 2024-04-27:
> dropping the pruned lines #~ ....

no, this are useful. In the https://lokalise.com/ it shows such old strings as a suggestion when you editing a translation. This helps a lot because you can simply click on them and change only that parts that was changed and needs to be re-translated.

Also sometimes you can simply recover the old translations like I did here:
https://github.com/openwrt/luci/pull/7090/commits/60461c07fffb66e7c5aa02804d7c6c204b06f0cc

The line numbers are needed too.

But if something that I would like to improve is:
1. Move the po files to a separate repo. So I can again subscribe to the repo changes without being annoyed by translation commits. This also will significantly clear history.
2. Get rid of all the translation scripts (e.g. i18n-sync.sh) : this must be a part of the gettext package or something like that. Maybe contribute them or find a replacement (GNOME or Qt must have something like that).
3. Make the Webblate to push directly without creating PRs that are waiting for long. This is probably can be achieved with the CODEOWNERS file with `*.po @weblate` line.
4. Maybe add more translation platforms e.g. Transifex or the mentioned above Lokalise. They may be less comfortable in some aspects but they have a very good AI translations that works very good from my experience. This can be any platform even a commercial because it's anyway just produces po files and we can change it any time.

closed 2024-05-27 by systemcrash