openwrt/luci #7091: LuCI allows (double-)quotes in firewall rule names, resulting in syntax error

mirko, 2024-04-28:
## Steps to reproduce:

1. go to: Network → Firewall → Traffic Rules
2. Create a rule with double-quotes in its name
3. Save & Apply
4. Everything *seems* to be fine

## Actual behavior:

1. Changes seem to be applied successfully, but are not
2. `/etc/init.d/firewall restart` resulting in: `/dev/stdin:77:99-102: Error: syntax error, unexpected string, expecting newline or semicolon - ip saddr 10.254.254.0/24 tcp dport { 80, 443 } counter accept comment "!fw4: Allow INPUT from \"MGMT\""`

## Expected behavior:
Double-quotes disallowed in LuCI webinterface for traffic rules, as resulting in syntax errors and/or error message when clicking Save & Apply

```
RA MTU and RA Hop Limit are displaying the correct values found in /etc/config/dhcp in black font color.
```

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
NAME="OpenWrt"
VERSION="23.05-SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 23.05-SNAPSHOT"
VERSION_ID="23.05-snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r23825-5cae98c25d"
OPENWRT_BOARD="x86/z51"
OPENWRT_ARCH="x86_64"
OPENWRT_TAINTS="no-all"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 23.05-SNAPSHOT r23825-5cae98c25d"

```

hnyman, 2024-04-28:
Yeah, it seems that the traffic rule has no validation datatype. 
https://github.com/openwrt/luci/blob/d4b8290f072767182e2d172da847291d49b9ce45/applications/luci-app-firewall/htdocs/luci-static/resources/view/firewall/rules.js#L209-L211

Possibly "uciname" would be right?
https://github.com/openwrt/luci/blob/d4b8290f072767182e2d172da847291d49b9ce45/modules/luci-base/htdocs/luci-static/resources/validation.js#L455-L457

Not sure if that would be too restrictive? 

The change below would implement that. Taken from live router:
```
--- /rom/www/luci-static/resources/view/firewall/rules.js
+++ /www/luci-static/resources/view/firewall/rules.js
@@ -209,6 +209,7 @@
                o = s.taboption('general', form.Value, 'name', _('Name'));
                o.placeholder = _('Unnamed rule');
                o.modalonly = true;
+               o.datatype = 'uciname';

                o = s.option(form.DummyValue, '_match', _('Match'));
                o.modalonly = false;
```

[image]

Ps. Being an old-school programmer used to "alhpanumerics only" in identifiers, I find it strange that somebody tries to use delimiter characters like quotes inside a name.

jow-, 2024-04-28:
Not a LuCI bug, and the name is a free formed text entry. Must be fixed in firewall4.

jow-, 2024-04-28:
So turns out that nftables syntax is really lame here, it's quoted string definition is literally `"[^"]*"`, see https://git.netfilter.org/nftables/tree/src/scanner.l#n126

So there's currently no way to have actual double quotes in double quoted strings. Since iptables did allow arbitrary characters and since the name field ends up as comment in the rules (so it has no effect on actual matching) I think that it makes sense to simply filter away double quotes (or silently substitute them with single quotes) in firewall4, this also ensures that existing uci firewall configs with such characters continue to work.

mirko, 2024-04-29:
> Ps. Being an old-school programmer used to "alhpanumerics only" in identifiers, I find it strange that somebody tries to use delimiter characters like quotes inside a name.

@hnyman OK, boomer ;) Maybe I'm a bit old-school, too, expecting things not to crash immediately when entering a "special character" :)

Jokes aside, personally I don't care too much, but the actual problem IMHO is, that no error is spit out to the regular user, while all firewall rules are being ignored due to a special character in a firewall rule name. Resulting syntax errors end up in a completely disabled firewall, without the user having any clue about it, leaving him/her into thinking the firewall and whatever rules is/are active, while it's not.

closed 2024-05-21 by openwrt-bot