openwrt/luci #7103: openclash

xxq240, 2024-05-01:
# Issue template:
1. Please make sure that the issue subject starts with `<package-name>:`
2. Issues related to end of life (EOL) releases are not supported or maintained and will be closed
3. Remove lines from top till here and fill in the following questions
***

## Steps to reproduce:
**Example: Replace the following lines and remove this**

1. go to: Network → Interfaces → LAN (an interface with a static address) → DHCP server tab → IPv6 RA settings tab
2. set RA MTU and RA Hop Limit to arbitrary non-default value
3. Save & Apply
4. go to: IPv6 RA settings tab again

## Actual behavior:
**Example: Replace the following lines and remove this**

1. option ra_mtu '1500' is set correctly in /etc/config/dhcp
2. In LuCI though, RA MTU and RA Hop Limit are still displaying the default values in light gray font color.

## Expected behavior:
**Example: Replace the following lines and remove this**

```
RA MTU and RA Hop Limit are displaying the correct values found in /etc/config/dhcp in black font color.
```

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

**Example: Replace the following lines and remove this**
```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='21.02.0-rc4'
DISTRIB_REVISION='r16256-2d5ee43dc6'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 21.02.0-rc4 r16256-2d5ee43dc6'
DISTRIB_TAINTS=''
```

dannil, 2024-05-01:
Looks like it's working properly for me on `23.05.2`, please upgrade to latest stable from `21.02.0-rc4` and try again, you're on x86 so no reason not to.

[image]

Mosney, 2024-05-13:
This issue is totally created by template, should be close/delete

closed 2024-05-13 by hnyman