openwrt/luci #7114: `luci-app-keepalived:` VRRP preempting option not being reflected on the configuration

grifos, 2024-05-09:
Please note: I'm really new to GitHub, I apologize for any mistakes.

Maintainer: if I'm not mistaken the maintainer should be Florian @feckert 

Environment: Belkin RT1800 (MT7621, mipsel_24kc), running OpenWrt 23.05.3. Same result when running OpenWrt snapshot r26235-0d74b2a1e5 from 2024-05-08.

I've also reported this on the forum: https://forum.openwrt.org/t/keepalived-vrrp-preempting-not-working/197133

## Steps to reproduce:
Two nodes, both starting in VRRP backup mode. The one meant to become the master (R1) has a priority of 120 and the other (R2) of 100. On the LAN VRRP instances, configure weighted tracking of the WAN interface with a weight of 80. When the WAN cable is disconnected at R1, R2 should become the master also for the LAN instances, preempting R1 due to its now higher priority, but that's not happening.

Below is an example instance from R2's `/tmp/keepalived.conf`

It seems that no matter what settings I change on the `luci-app-keepalived` (specifically the `Disable Preempt` option is turned off, but I've tried various other combinations of settings), the `nopreempt` line never goes away from the configuration.

```
vrrp_instance iot {
	authentication {
		auth_type PASS
		auth_pass xxxxxxxx
	}
	state BACKUP
	interface br-lan.501
	unicast_src_ip 192.168.101.253
	virtual_router_id 151
	version 3
	priority 100
	advert_int 5
	preempt_delay 30
	garp_master_delay 1
	garp_master_refresh 1
	garp_master_repeat 1
	garp_master_refresh_repeat 1
	nopreempt
	notify_backup "/bin/busybox env -i ACTION=NOTIFY_BACKUP TYPE=INSTANCE NAME=iot /sbin/hotplug-call keepalived"
	notify_master "/bin/busybox env -i ACTION=NOTIFY_MASTER TYPE=INSTANCE NAME=iot /sbin/hotplug-call keepalived"
	notify_fault "/bin/busybox env -i ACTION=NOTIFY_FAULT TYPE=INSTANCE NAME=iot /sbin/hotplug-call keepalived"
	notify_stop "/bin/busybox env -i ACTION=NOTIFY_STOP TYPE=INSTANCE NAME=iot /sbin/hotplug-call keepalived"
	virtual_ipaddress {
		192.168.101.254 dev br-lan.501 label br-lan.501:v501 scope global
	}
	track_interface {
		wan weight 80
	}
	unicast_peer {
		192.168.101.251
	}
}
```
The logs also report that VRRP is running in `nopreempt` mode.

```
Thu May  9 12:37:01 2024 daemon.info Keepalived_vrrp[3360]: (wan) preempt_delay is incompatible with nopreempt mode - resetting
Thu May  9 12:37:01 2024 daemon.info Keepalived_vrrp[3360]: (lan) preempt_delay is incompatible with nopreempt mode - resetting
Thu May  9 12:37:01 2024 daemon.info Keepalived_vrrp[3360]: (iot) preempt_delay is incompatible with nopreempt mode - resetting
```

## Expected behavior:
Being able to run VRRP in `preempt` mode, so that the node with the highest priority would become master.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.3'
DISTRIB_REVISION='r23809-234f1a2efa'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 23.05.3 r23809-234f1a2efa'
DISTRIB_TAINTS=''
```

feckert, 2024-05-10:
@grifos Thank you for your bug report. Should be fixed.

closed 2024-05-10 by feckert

grifos, 2024-05-10:
Thank you so much Florian. Up to you if you want to backport it, though it would probably just be for one user, eg. me, then I can easily just upgrade to snapshot.

commit 7656c64a (2024-05-10): luci-app-keepalived: fix nopreempt value handling

The 'nopreempt' option is marked with the prefix 'no_value_' in the
keepalived init script. This means that if the value is not set in the
keepalived configuration, it must be removed from the uci configuration.

To get this behavior, the rmempty flag in the LuCI must be removed.

fixes #7114