openwrt/luci #7123: luci-app-p910nd: server is effectively always disabled

thebodzio, 2024-05-13:
## Steps to reproduce:

1. Go to Services → p910nd Printer server.
2. Edit and disable previously enabled printer.
3. Save and apply the changes.
4. Edit the printer again and enable it.
5. Save and apply the changes.

## Actual behavior:

1. LuCI shows the printer as enabled.
2. `enabled` option is removed from the `/etc/config/p910nd` section corresponding to the printer.
3. `p910nd` server for the given printer is _not_ running.

## Expected behavior:

Whenever printer is enabled in LuCI, `enable '1'` option should be placed in the appropriate section of the `/etc/config/p910nd` config file and the server should be started.

## Additional Information:

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.0'
DISTRIB_REVISION='r23497-6637af95aa'
DISTRIB_TARGET='bcm27xx/bcm2708'
DISTRIB_ARCH='arm_arm1176jzf-s_vfp'
DISTRIB_DESCRIPTION='OpenWrt 23.05.0 r23497-6637af95aa'
DISTRIB_TAINTS=''
```

## Observations

When printer is being enabled, “Unsaved changes” show that `enable` option is to be removed:

[image]

This, however, causes the server to be effectively disabled, since [/etc/init.d/p910nd](https://github.com/openwrt/packages/blob/4a4090ff89bb9ef45e5d61339ca1f143e6997f24/net/p910nd/files/p910nd.init#L32-L38) starts the server only if both `enable` option is > 0 and appropriate device exists:

```
if [ "$enabled" -gt 0 ] && [ -e "$device" ]; then
```

Since `enable` option is removed completely init script gets:

```
config_get_bool "enabled" "$section" "enabled" '0'
```

In other words: server is _disabled_ and not _enabled_ by default.

systemcrash, 2024-05-14:
Good find! Thanks for pointing this out. Resolved in 05c6a0887f88dd65b8176686f3aee5bd05d8771a.

systemcrash, 2024-05-14:
Cherry-picked to 23 and 22 branches.

closed 2024-05-14 by systemcrash

hnyman, 2024-05-14:
@systemcrash 

I am not at all sure that the fix is correct... 

The fix may mask the mismatch, but looking at the underlying package, the default value for 'enabled' is 0 while LuCI thinks that it is 1 and has (so far) removed 1 as unnecessary:
https://github.com/openwrt/packages/blob/f805121318acb6d54fa3b5d80b2ba8d90e40c427/net/p910nd/files/p910nd.config#L7-L7

More correct fix would be to adjust LuCI's default to the actual package.

hnyman, 2024-05-14:
@systemcrash 
I have to apologize. 
Seems you actually did quite right ;-)

You added the rmempty statement (which is the possible "mask" part my comment was based on), 
but you also removed the erroneous default = true, and that implicitly leads to default=false, which is correct, and which I overlooked.

Looking at the final code, I think that it is ok. 
Disabled is the implicit default and rmempty guarantees that LuCI does not remove enabled in any circumstance.

https://github.com/openwrt/luci/blob/05c6a0887f88dd65b8176686f3aee5bd05d8771a/applications/luci-app-p910nd/htdocs/luci-static/resources/view/p910nd.js#L107-L108

Sorry for the noise.

systemcrash, 2024-05-14:
Don't worry. It's good to double check. Those settings are responsible for
some confusion 😅

thebodzio, 2024-05-14:
Thank you very much for such a rapid fix! 👏
Your efforts are most appreciated, sir! 🎩👌😊

jow-, 2024-05-15:
Please remove the `rmempty = false` policy is to only write options having non-standard values.