openwrt/luci #7134: luci-app-ocserv breaks ocserv on 23.05.03

buboleck, 2024-05-22:
## Steps to reproduce:

1. Install Ocserv via package manager
2. install luci-app-ocserv via package manager
3. Apply Ocserv configuration via the web interface.
4. Try to connect to the Ocserv instance.
5.  Ocserv crashes  and is not usable at all with the below message in the logs:
Mon May 13 11:38:46 2024 daemon.info procd: Instance ocserv::instance1 s in a crash loop 6 crashes, 0 seconds since last crash

## Actual behavior:
Ocserv crashes instead of receiving proper config.

## Expected behavior:
No crash and proper working config.

## Additional Information:
The crash is reproducible with the following combinations:

OpenWrt 23.05.02/Ocserv 1.3/luci-app-ocserv git-20.110.55046-74da73b
OpenWrt 23.05.03/Ocserv 1.3/luci-app-ocserv git-20.110.55046-74da73b
OpenWrt SNAPSHOT r26300 (Linux 6.6.30)/Ocserv 1.3/luci-app-ocserv git-20.110.55046-74da73b

The only working combination for me is OpenWrt 23.05.02/Ocserv 1.1.x/1.2.x/luci-app-ocserv git-20.110.55046-74da73b

OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.2'
DISTRIB_REVISION='r23630-842932a63d'
DISTRIB_TARGET='mvebu/cortexa9'
DISTRIB_ARCH='arm_cortex-a9_vfpv3-d16'
DISTRIB_DESCRIPTION='OpenWrt 23.05.2 r23630-842932a63d'
DISTRIB_TAINTS=''
```
Please let me know what other info I can provide.

systemcrash, 2024-05-22:
Seems like an ocserv package problem. If it or its startup script cannot handle parameters, then the bug is there. Open an issue in the respective repo where ocserv is found. Provide info of the factory default config and the config produced after using this app.

buboleck, 2024-05-22:
I just tested ocserv-1.1.6 on OpenWrt SNAPSHOT r26300 and it works, so definitely ocserv related.

closed 2024-05-22 by buboleck