openwrt/luci #7142: [To stop stupid users from bricking hardware – Suggestion] Disallow „*ipk“ from being flashed.

dziugas1959, 2024-05-29:
Ok, so I will use this as context and reason, one of my friends, had installed „Openwrt“ and wanted to install a theme from „Github“ (quite popular in fact), the critical mistake that he made, was that he had the „Linus Tech Tips/Linus Sebastian“ mindset, where he ignored all the red text and warning and flashed the „*.ipk“ file and bricked his device.

1. go to: Backup and Flash firmware – flash firmware – ignore warning – brick device.

So my suggestion is, that when uploading „opkg“ type files (in this case *.ipk), redirect the user to „opkg“ package manager and it's archive upload.
About other formats, no comment, since I would not expect users to make other mistakes or install other binaries, which is not in the scope of this suggestion.

jow-, 2024-05-29:
Not possible, .ipk archives are gzipped tar archives, some sysupgrade files are gzipped tar files as well. Plus there already is a big fat integrity warning.

closed 2024-05-29 by jow-

dziugas1959, 2024-05-29:
> Not possible, .ipk archives are gzipped tar archives, some sysupgrade files are gzipped tar files as well. Plus there already is a big fat integrity warning.

As I said, not to mess with zips or tar archives, just „opkg“ packages, if a person wants to flash with „*.ipk“, then they should use the console.

jow-, 2024-05-29:
The point is that, from a program point of view, it is not easy to distinguish a sysupgrade file from an opkg package. Both start with a gzip header, both contain a tar archive. You would need to apply heuristics based on the content of the archive, which means unpacking it etc.

This is a lot of effort for something that already is convered by generic warnings. Yes, stock firmware might be able to do this because their firmware images have distinct headers (and likely no package support at all).

jow-, 2024-05-29:
1) Extension checks are unreliable, files could be arbitrarily named
2) At the point where the code runs, the original file name is not even known anymore, a file uploaded from the flash firmware page will be stored as `/tmp/firmware.bin` internally

dziugas1959, 2024-05-29:
> 1. Extension checks are unreliable, files could be arbitrarily named
> 
>     2. At the point where the code runs, the original file name is not even known anymore, a file uploaded from the flash firmware page will be stored as `/tmp/firmware.bin` internally

Agreed and agreed, but this is a check for stupid people who don't realize that there are two places to upload archives. It doesn't need much, the „LuCI“ interface asks for a file/archive, you give it, if extension is matched as „opkg“ format, then redirect.

dziugas1959, 2024-06-03:
Okay, another suggestion if that is too much, how about:

If the check fails, then ask for the root password.

stangri, 2024-06-03:
> 1. Extension checks are unreliable, files could be arbitrarily named

@jow- I think it's a pretty low risk innovation to prevent the .ipk/.apk files from being flashed as a firmware upgrade. In the world of Android/magisk you do legitimately "flash" some third-party packages, so I see a line of thinking that flashing an OpenWrt package same way is a way to go. 

AFAIR, the generic code checking for the firmware upgrade eligibility was part of the openwrt/openwrt repo, if I make a PR there, would you be willing to accept that?