openwrt/luci #7155: mod-status: please add more keywords to view/nftables.js

brada4, 2024-06-11:
Status/firewall may need to parse `notrack` - similar to `accept` and `goto` similar to `jump` but ends after called chain.

systemcrash, 2024-06-13:
Is this something fw4 itself handles also?

brada4, 2024-06-13:
I have intent to make it do so around handle_reset.

brada4, 2024-11-30:
After half year of intense translation notrack is still translated in error box
And in 2410 nothing gets parsed.

closed 2024-11-30 by systemcrash

brada4, 2024-11-30:
it is vmap there too, it is a bit tricky, it is like true false map but with multiple outcomes in parentheses.

[image]

brada4, 2024-12-01:
I dont get your sarcasm. Please instrument what comes out of this PR
https://github.com/openwrt/firewall4/pull/22
And reopen issue and avoid snarky comments.