openwrt/luci #7159: luci-app-strongswan-swanctl: Encryption proposal disappears when adding tunnel configuration

mikma, 2024-06-15:
## Steps to reproduce:

1. go to: VPN → strongSwan Configuration → Encryption Proposal
2. enter "vpn1" and click on "Add".
3. select "ESP Proposal".
4. Save & Apply
5. go to: Tunnel Configuration tab
6. enter "vpn1" and click on "Add"
7. select the crypto proposal "vpn1". Enter any local and remote subnet.
8. Save & Apply
9. go to: Encryption Proposal tab again.

## Actual behavior:

1. The "vpn1" line in Encryption Proposal is gone.

## Expected behavior:

1. The "vpn1" line in Encryption Proposal should exist.

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r26515-6e51b363e9'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r26515-6e51b363e9'
DISTRIB_TAINTS=''
```

This problem is related to [Named uci sections may be overwritten by misstake in luci](https://forum.openwrt.org/t/named-uci-sections-may-be-overwritten-by-misstake-in-luci/200130)

systemcrash, 2024-07-08:
@jempatel could you take a look at this? I've verified this behaviour. 

Needs an error check to prevent in-use names from being consumed by a new definition, for each of the section types in use.

tkkost, 2025-07-19:
I have the same error. How can I setup an ipsec tunnel?

commit eb66bdcc (2025-07-19): luci-app-strongswan-swanctl: prevent section name collisions

Named sections collide and overwrite each other when navigating the
GUI and repeating the same name, because uci does not allow multiple
named sections to share the same name. This at least forces users to
pick names so their resulting config works.

Closes #7159
commit 92b672b3 (2025-07-19): luci-app-strongswan-swanctl: prevent section name collisions

Named sections collide and overwrite each other when navigating the
GUI and repeating the same name, because uci does not allow multiple
named sections to share the same name. This at least forces users to
pick names so their resulting config works.

Closes #7159

(cherry picked from commit eb66bdccf815797f6d1fa0cc0aed571319a2b980)
closed 2025-07-19 by systemcrash