openwrt/luci #7171: Fix this single/lone read-only string.

dziugas1959, 2024-06-20:
I already had mentioned this, but seems like no one paid any attention to it, this single string is read-only. 
[image]

Other languages do have it translated before it was set like that, others like Russian and Lithuanian don't. 
[image]

„Weblate“ counts it as another category.
[image]

Such strings should not appear in a translation project if they are not supposed to be translated, but in this case, it should be translated, as in Lithuanian (žr. 2019-11-07 Valstybinės lietuvių kalbos komisijos nutarimą Nr. N-8 (178) „Dėl Lietuvių kalbos skyrybos taisyklių“) it would be:
„Webroot“
And Russian:
«Webroot»

dziugas1959, 2024-06-20:
P.S. Red ! 
[image]
[image]
[image]
Only a couple of components seem to be affected.

systemcrash, 2024-06-21:
I don't know about weblate, but a manual PR here would work :)

dziugas1959, 2024-06-24:
> I don't know about weblate, but a manual PR here would work :)

A simple „flag“ removal/edit to editable of the string would work.
[image]

This can only be done by maintainers.

hnyman, 2024-06-25:
I removed the read-only flag in weblate from that string in Lithuanian, but no idea how that flag has surfaced there.

closed 2024-06-25 by hnyman