openwrt/luci #7183: [luci-app-mwan3] Cannot add multi IP rules.

marianomd, 2024-07-04:
Currently it's not possible to specify multiple source/destination IP addresses in luci-app-mwan3:
[image]

## Steps to reproduce:

1. go to: Network → MultiWAN Manager → Rule tab
2. add or edit a rule
3. try to specify more than one IP address in destination

## Actual behavior:

1. The destination IP input becomes red and it's not possible to save.

## Expected behavior:

Allow saving.

## Additional Information:

LuCI openwrt-23.05 branch (git-24.086.45142-09d5a38)

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.3'
DISTRIB_REVISION='r23809-234f1a2efa'
DISTRIB_TARGET='ramips/mt7621'
DISTRIB_ARCH='mipsel_24kc'
DISTRIB_DESCRIPTION='OpenWrt 23.05.3 r23809-234f1a2efa'
DISTRIB_TAINTS=''
```

systemcrash, 2024-07-08:
@feckert want to have a crack at fixing?

feckert, 2024-07-08:
That´s because of the following lines.
https://github.com/openwrt/luci/blob/master/applications/luci-app-mwan3/htdocs/luci-static/resources/view/mwan3/network/rule.js#L54
and 
https://github.com/openwrt/luci/blob/master/applications/luci-app-mwan3/htdocs/luci-static/resources/view/mwan3/network/rule.js#L63
LuCI wants to have an IP address here and not a comma-separated list of IP Addresses.

systemcrash, 2024-07-08:
mwan3 itself should not be able to handle multiple IPs?

closed 2024-07-08 by systemcrash