openwrt/luci #7186: luci-mod-status: syslog scroll buttons inert on iOS Safari browser

dave14305, 2024-07-07:
## Steps to reproduce:

1. go to: Status → System Log using an iOS/iPadOS Safari browser with LuCI Bootstrap theme
2. Click the Scroll to tail button. Nothing happens.
3. Scroll manually to the bottom of the page. Click the Scroll to head button. Nothing happens.
4. Repeat same steps on the Kernel log page.

## Actual behavior:

1. No scrolling

## Expected behavior:

1. Scrolling to bottom or top of page as worked earlier in 23.05. Broken with https://github.com/openwrt/luci/commit/9370bdddaede2feeb581193158d83f5062d5a318

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.3'
DISTRIB_REVISION='r23809-234f1a2efa'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt 23.05.3 r23809-234f1a2efa'
DISTRIB_TAINTS=''
```

systemcrash, 2024-07-08:
I can at least confirm this behaviour on iOS Safari. @McGiverGim wanna take a crack at fixing?

McGiverGim, 2024-07-08:
I don't have any mac/apple product. I will be out of my computer for one month more or less. I can give it a try then. I will need to install safari on my windows to reproduce it?

dave14305, 2024-07-08:
Consider using `.scrollIntoView()` instead of `.focus()`. Seems to work on iOS Safari and Windows Chrome for the current BootStrap theme.

systemcrash, 2024-07-08:
@dave14305 your suggestion worked fine. Committed. Thanks!

closed 2024-07-08 by systemcrash

dave14305, 2024-07-30:
> @dave14305 your suggestion worked fine. Committed. Thanks!

@systemcrash will this be backported to 23.05?

systemcrash, 2024-07-30:
If it wasn't already I'll go do that. Good to get it verified by others in
master (where verified means it doesn't break anything else)

systemcrash, 2024-07-30:
picked to 23 and 22. Thanks for the nudge @dave14305

commit 7893f2d0 (2024-07-08): luci-mod-status: log viewers - change focus() to scrollIntoView()

scrollIntoView() is more universally supported and standards tracked.

Closes #7186

Updates 9370bdddaede2feeb581193158d83f5062d5a318
commit 910cf4da (2024-07-30): luci-mod-status: log viewers - change focus() to scrollIntoView()

scrollIntoView() is more universally supported and standards tracked.

Closes #7186

Updates 9370bdddaede2feeb581193158d83f5062d5a318

(cherry picked from commit 7893f2d08aaa1a0419ede1a695a25781f60e6ba6)
commit b6da3f27 (2024-07-30): luci-mod-status: log viewers - change focus() to scrollIntoView()

scrollIntoView() is more universally supported and standards tracked.

Closes #7186

Updates 9370bdddaede2feeb581193158d83f5062d5a318

(cherry picked from commit 7893f2d08aaa1a0419ede1a695a25781f60e6ba6)