openwrt/luci #7193: luci-mod-network: wireless: WPA3 Enterprise config does not match what luci shows

nasbdh9, 2024-07-13:
https://github.com/openwrt/netifd/commit/db3934d2f740bdfe8537933741f71b439a109422

missing `wpa3-192` = `WPA3 802.1X (GCMP-256)`
option encryption `wpa3` is actually `wpa2+ccmp` = `WPA2 802.1X (CCMP)`

nasbdh9, 2024-07-14:
link https://github.com/openwrt/openwrt/issues/15347

systemcrash, 2024-07-15:
## Requires wpa-supplicant with EAP Suite-B support.

You'll need to make a PR of what you want changing, pending the outcome of discussion (if there ever is one) in the linked issue.

Luci has functionality as: `has_ap_eap192` for WPA3. This check is:

```js
					// Probe Suite-B support
					var has_ap_eap192 = L.hasSystemFeature('hostapd', 'suiteb192'),
					    has_sta_eap192 = L.hasSystemFeature('wpasupplicant', 'suiteb192');
```

which is checked by UCI at runtime:

```js
			const wifi_features = [ 'eap', '11ac', '11ax', '11r', 'acs', 'sae', 'owe', 'suiteb192', 'wep', 'wps' ];

			if (access('/usr/sbin/hostapd')) {
				result.hostapd = { cli: access('/usr/sbin/hostapd_cli') == true };

				for (let feature in wifi_features)
					result.hostapd[feature] = system(`/usr/sbin/hostapd -v${feature} >/dev/null 2>/dev/null`) == 0;
			}

			if (access('/usr/sbin/wpa_supplicant')) {
				result.wpasupplicant = { cli: access('/usr/sbin/wpa_cli') == true };

				for (let feature in wifi_features)
					result.wpasupplicant[feature] = system(`/usr/sbin/wpa_supplicant -v${feature} >/dev/null 2>/dev/null`) == 0;
			}
```

Check yourself, on an openwrt shell:
```sh
hostapd -vsuiteb192
echo $?
0
```

nasbdh9, 2024-07-16:
`/etc/config/wireless` → `option encryption 'wpa2+ccmp'` → luci show as `WPA2 802.1X (CCMP)`

<details>
<summary>option encryption 'wpa2+ccmp'</summary>
<pre><code>
interface=phy1-ap0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
auth_server_addr=127.0.0.1
auth_server_port=1812
auth_server_shared_secret=
dynamic_own_ip_addr=1
eapol_key_index_workaround=1
ieee8021x=1
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=Test2.2.1
bridge=br-lan
wds_bridge=
snoop_iface=br-lan.1000
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-EAP WPA-EAP-SHA256
okc=0
disable_pmksa_caching=1
ieee80211w=2
group_mgmt_cipher=AES-128-CMAC
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
bssid=
#default_macaddr
</code></pre>
</details>

`/etc/config/wireless` → `option encryption 'wpa3+ccmp'` → luci show as `WPA2 802.1X (CCMP)` ← show wrong

<details>
<summary>option encryption 'wpa3+ccmp'</summary>
<pre><code>
interface=phy1-ap0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
sae_require_mfp=1
sae_pwe=2
auth_server_addr=127.0.0.1
auth_server_port=1812
auth_server_shared_secret=UXOYL#1@MwV^JQq!XRW2MNaFgL@j#NCs
dynamic_own_ip_addr=1
eapol_key_index_workaround=1
ieee8021x=1
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=Test2.2.1
bridge=br-lan
wds_bridge=
snoop_iface=br-lan.1000
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-EAP-SHA256
okc=0
disable_pmksa_caching=1
ieee80211w=2
group_mgmt_cipher=AES-128-CMAC
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
bssid=d8:84:66:c0:7e:38
#default_macaddr
</code></pre>
</details>

`option encryption 'wpa2+ccmp'` compared `option encryption 'wpa3+ccmp'`
add `sae_require_mfp=1 sae_pwe=2`
change `wpa_key_mgmt=WPA-EAP WPA-EAP-SHA256` to`wpa_key_mgmt=WPA-EAP-SHA256`

nasbdh9, 2024-07-16:
### **wpa2+ccmp**: 

`/etc/config/wireless` → `option encryption 'wpa2+ccmp'` → luci show as `WPA2 802.1X (CCMP)`

<details>
<summary>option encryption 'wpa2+ccmp'</summary>
<pre><code>
interface=phy1-ap0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
auth_server_addr=127.0.0.1
auth_server_port=1812
auth_server_shared_secret=
dynamic_own_ip_addr=1
eapol_key_index_workaround=1
ieee8021x=1
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=Test2.2.1
bridge=br-lan
wds_bridge=
snoop_iface=br-lan.1000
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-EAP WPA-EAP-SHA256
okc=0
disable_pmksa_caching=1
ieee80211w=2
group_mgmt_cipher=AES-128-CMAC
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
bssid=
#default_macaddr
</code></pre>
</details>

### **wpa3+ccmp**: 

`/etc/config/wireless` → `option encryption 'wpa3+ccmp'` → luci show as `WPA2 802.1X (CCMP)` ← show wrong

<details>
<summary>option encryption 'wpa3+ccmp'</summary>
<pre><code>
interface=phy1-ap0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
sae_require_mfp=1
sae_pwe=2
auth_server_addr=127.0.0.1
auth_server_port=1812
auth_server_shared_secret=
dynamic_own_ip_addr=1
eapol_key_index_workaround=1
ieee8021x=1
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=Test2.2.1
bridge=br-lan
wds_bridge=
snoop_iface=br-lan.1000
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-EAP-SHA256
okc=0
disable_pmksa_caching=1
ieee80211w=2
group_mgmt_cipher=AES-128-CMAC
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
bssid=
#default_macaddr
</code></pre>
</details>

`option encryption 'wpa2+ccmp'` compared `option encryption 'wpa3+ccmp'`
add `sae_require_mfp=1 sae_pwe=2`
change `wpa_key_mgmt=WPA-EAP WPA-EAP-SHA256` to `wpa_key_mgmt=WPA-EAP-SHA256`

### **wpa3-192:**

`/etc/config/wireless` → `option encryption 'wpa3-192'` → luci show as `WPA3 802.1X (GCMP-256)` ← misleading, luci it should show `WPA3 802.1X 192-bit mode (CNSA)`, lacks the option to configure `/etc/config/wireless` to `option encryption 'wpa3-192'`

<details>
<summary>option encryption 'wpa3-192'</summary>
<pre><code>
interface=phy1-ap0
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=1
skip_inactivity_poll=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
sae_require_mfp=1
sae_pwe=2
auth_server_addr=127.0.0.1
auth_server_port=1812
auth_server_shared_secret=
dynamic_own_ip_addr=1
eapol_key_index_workaround=1
ieee8021x=1
auth_algs=1
wpa=2
wpa_pairwise=GCMP-256
ssid=Test2.2.1
bridge=br-lan
wds_bridge=
snoop_iface=br-lan.1000
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-EAP-SUITE-B-192
okc=0
disable_pmksa_caching=1
ieee80211w=2
group_mgmt_cipher=BIP-GMAC-256
qos_map_set=0,0,2,16,1,1,255,255,18,22,24,38,40,40,44,46,48,56
bssid=
#default_macaddr
</code></pre>
</details>

`option encryption 'wpa3+ccmp'` compared `option encryption 'wpa3-192'`
change1 `wpa_pairwise=CCMP` to `wpa_pairwise=GCMP-256`
change2 `wpa_key_mgmt=WPA-EAP-SHA256` to `wpa_key_mgmt=WPA-EAP-SUITE-B-192`
change3 `group_mgmt_cipher=AES-128-CMAC` to `group_mgmt_cipher=BIP-GMAC-256`

### **other**:

`/etc/config/wireless` → `option encryption 'wpa2+ccmp256'`
`/etc/config/wireless` → `option encryption 'wpa2+gcmp256'`
`/etc/config/wireless` → `option encryption 'wpa3+ccmp256'`
`/etc/config/wireless` → `option encryption 'wpa3+gcmp256'`
These four options can indeed make hostapd work, but I can't find any document that considers the above four modes to be standard, and I can't connect to the SSID set with these four modes in Android and iOS.

systemcrash, 2024-07-16:
You can try [my commit](https://github.com/systemcrash/luci/commit/915b3c5ad1091b23ff86ed557c9869061df77caf) and see whether it resolves your usability problem.

nasbdh9, 2024-07-16:
> You can try [my commit](https://github.com/systemcrash/luci/commit/915b3c5ad1091b23ff86ed557c9869061df77caf) and see whether it resolves your usability problem.

apply the commit wpa3-192 can be configured in luci.

nasbdh9, 2024-07-16:
> > luci show as `WPA2 802.1X (CCMP)` ← show wrong, luci it should show `WPA3 802.1X (CCMP)`
> 
> Then a correction must be made in the underlying scripts as @Ansuel mentions [here](https://github.com/openwrt/openwrt/issues/15347#issuecomment-2094870564), if there is indeed even a bug there. Because that is where the data comes from.
> 
> https://github.com/openwrt/openwrt/blob/b4f1deab9041583e7c350df4a490f6c8d999614b/package/network/config/wifi-scripts/files/lib/netifd/netifd-wireless.sh#L260
> 
> https://github.com/openwrt/openwrt/blob/b4f1deab9041583e7c350df4a490f6c8d999614b/package/network/config/wifi-scripts/files/lib/netifd/hostapd.sh#L49
> 
> But I don't think there's a bug (if there is, remind me):
> 
I will open an issue in netifd.

systemcrash, 2024-07-16:
> > You can try [my commit](https://github.com/systemcrash/luci/commit/915b3c5ad1091b23ff86ed557c9869061df77caf) and see whether it resolves your usability problem.
> 
> apply the commit wpa3-192 can be configured in luci.

Does this fix config issues for you? You set wpa-192 and your AP sets the correct crypto?

nasbdh9, 2024-07-16:
Yes, after apply the commit and configuring `WPA3-EAP 192-bit Mode` in luci, hostapd runs correctly in wpa3-192 mode.
But when configuring `wpa3+ccmp`, luci still shows `WPA2 802.1X (CCMP)`, and check `iwinfo` also show `WPA2 802.1X (CCMP)`, where should report this issues?

systemcrash, 2024-07-17:
https://github.com/openwrt/iwinfo ?

systemcrash, 2024-07-17:
OK - the fix is in. 

Thanks for the detective work @nasbdh9

closed 2024-07-17 by systemcrash

damnms, 2026-08-19:
guess i have the same problem, its shown as wpa2 but should be wpa3 - running openwrt main/master self compiled every 1-2 weeks
[image]

commit 3bf3c0a0 (2024-07-17): luci-mod-network: Present wpa3-192bit option in wifi

Closes #7193

See https://github.com/openwrt/netifd/commit/db3934d2f740bdfe8537933741f71b439a109422