openwrt/luci #7196: VLAN config not working properly

palbiez, 2024-07-15:
When I'm creating 2 new VLANs in 23.05.3 on my Cudy WR1300 v3 all is working fine in the first view. 
I change the bridge LAN and add VLAN 1 untagged all ports also VLAN 140 tagged all ports for Guest WiFi with multiple APs. 
The configuration is looking fine. 
Relating to https://github.com/openwrt/luci/issues/5948 
I saw that vlan_filtering is set to 0. 
```

# cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option packet_steering '1'
        option ula_prefix 'fd9f:f265:6dcf::/48'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth0'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        option vlan_filtering '0'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.130.1'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option gateway '192.168.2.1'
        list dns '8.8.8.8'

config interface 'wan'
        option device 'wan'
        option proto 'static'
        option ipaddr '192.168.2.2'
        option gateway '192.168.2.1'
        option netmask '255.255.255.0'
        list dns '8.8.8.8'
        list dns '8.8.4.4'

config interface 'wan6'
        option device 'wan'
        option proto 'dhcpv6'

config route
        option interface 'lan'
        option target '0.0.0.0/0'
        option gateway '192.168.2.2'

config route
        option interface 'wan'
        option target '0.0.0.0/0'
        option gateway '192.168.2.1'

config device
        option name 'phy1-ap0'

config interface 'Gast'
        option proto 'static'
        option device 'br-lan.140'
        option ipaddr '192.168.140.1'
        option netmask '255.255.255.0'

config bridge-vlan
        option device 'br-lan'
        option vlan '1'
        list ports 'eth0:u*'
        list ports 'lan1:u*'
        list ports 'lan2:u*'
        list ports 'lan3:u*'
        list ports 'lan4:u*'

config bridge-vlan
        option device 'br-lan'
        option vlan '140'
        list ports 'eth0:t'
        list ports 'lan1:t'
        list ports 'lan2:t'
        list ports 'lan3:t'
        list ports 'lan4:t'

```
But if I add one or more VLAN to this bridge the vlan_filtering tag has been deleted by uci. 
And my network device crashes. 
[image]

[image]

[image]

I think the deletion of vlan_filtering is wrong and it should be still vlan_filtering 0 
In second step the vlans should be activated with a separate activation button or something like this.

systemcrash, 2024-07-15:
Try it without VLAN 1 - https://community.cisco.com/t5/switching/what-is-difference-between-default-vlan-and-native-vlan/td-p/2095204 - and see if you get the same result.

palbiez, 2024-07-15:
I tested today evening only on another device. 
```

root@DAP1325_terasse:~# cat /etc/config/network

config interface 'loopback'
        option device 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fd5e:4129:8e90::/48'

config device
        option name 'br-lan'
        option type 'bridge'
        list ports 'eth0'

config interface 'lan'
        option device 'br-lan'
        option proto 'static'
        option ipaddr '192.168.130.244'
        option netmask '255.255.255.0'
        option gateway '192.168.130.1'
        list dns '192.168.130.1'
        list dns '192.168.2.1'

config device
        option type 'bridge'
        option name 'Br-test'
        list ports 'eth0'

config bridge-vlan
        option device 'Br-test'
        option vlan '130'
        list ports 'eth0:t*'

config bridge-vlan
        option device 'Br-test'
        option vlan '140'
        list ports 'eth0:t'

config bridge-vlan
        option device 'Br-test'
        option vlan '150'
        list ports 'eth0:t'

config bridge-vlan
        option device 'Br-test'
        option vlan '128'
        list ports 'eth0:t'

config interface 'ALBkinder'
        option proto 'static'
        option device 'Br-test.140'
        option ipaddr '192.168.130.244'
        option netmask '255.255.255.0'

```
Deleted VLAN1 and made VLAN130 as default 
[image]
Used this interface (Br-Test) at the adapter. 
[image]

I must wait 90 seconds and then it will be reverted. 
When I'm deleting the primary VLAN and use it again. 
[image]
I must also wait 90 seconds and the interface will be reverted. 

Test it tomorrow onto main router. 
Should another VLAN made as default or no default VLAN should be defined when I'm not using VLAN 1?

palbiez, 2024-07-15:
I haven't applied the config yet but when I'm deleting VLAN1 and add another one the vlan_filtering will be deleted completely. 
[image]
I can apply this tomorrow morning and test again but I think the vlan_filtering tag shouldn't be deleted here

palbiez, 2024-07-16:
Same result without VLAN 1 
When I want to delete VLAN 1 uci want to delete the vlan_filtering tag. 
Also I'm not able to do this because my router is hanging then. I must wait 90 seconds then the changes will be reverted. 
Have been tested the following configurations. 
[image]
Deletion of VLAN 1 all other options will be the same.

[image]
Deletion of VLAN1 and add another tagged VLAN.

Edited VLAN ID from 1 to 130
[image]

In all cases the VLAN filtering tag want to be deleted by luci

palbiez, 2024-07-16:
Second test. 
Add a new bridge with ports Lan2-4 
Adding 2 VLANs 130 and 140.
[image]
[image]
No VLAN_filtering tag will be set by luci. What is right because the tag should be set at the interface I think.

```

config device
        option type 'bridge'
        option name 'Br-test'
        list ports 'eth0'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'

config bridge-vlan
        option device 'Br-test'
        option vlan '130'
        list ports 'eth0:t*'
        list ports 'lan1:t*'
        list ports 'lan2:t*'
        list ports 'lan3:t*'
        list ports 'lan4:t*'

config bridge-vlan
        option device 'Br-test'
        option vlan '140'
        list ports 'eth0:t'
        list ports 'lan1:t'
        list ports 'lan2:t'
        list ports 'lan3:t'
        list ports 'lan4:t'

```
Added new interface in Luci 
[image]
The VLAN filtering option won't be set here. 
```

config interface 'Management'
        option proto 'static'
        option device 'Br-test'
        option ipaddr '192.168.128.1'
        option netmask '255.255.255.0'

root@WR1300v3:~#
```

I'm very confused at the moment when the vlan_filtering tag should be set and when it should be set from 0 to 1

palbiez, 2024-07-16:
Last test.
Deleted VLAN 1 from the config directly in /etc/config/network
And then change anything at the VLAN configuration in luci. 
Luci want to delete VLAN filtering tag here again. 
[image]

palbiez, 2024-07-16:
The issue contains with the VLAN filtering checkbox. 
If you set filtering checkbox but you don't define any vlans the vlan_filtering tag is set to 1
[image]
If you define a VLAN without checking this checkbox manually VLAN filtering tag won't be set by luci 
[image]

I think in this case the checkbox shouldn't be set and the tag should set to 0

Uncheck the box manually the tag will set to 0
[image]

I only saved this config and don't apply it. 
When I'm adding now the second VLAN 
The tag will be deleted accidentally I think it should be still on 0 and the checkbox shouldn't be checked
[image]

systemcrash, 2024-07-16:
That `vlan_filtering` is removed is normal. The internal default is 0, which is synonymous with the setting being removed. 

>  And my network device crashes.

Do you mean you get locked out and the device stops responding? This is because you lose network access: the device starts sending tagged frames as soon as you apply the config.

palbiez, 2024-07-16:
> That `vlan_filtering` is removed is normal. The internal default is 0, which is synonymous with the setting being removed.
Can you please tell me why the vlan_filtering has been deleted when I'm adding a new  VLAN?
I thought that vlan_filtering must be set otherwise it isn't working
 
> Do you mean you get locked out and the device stops responding? This is because you lose network access: the device starts sending tagged frames as soon as you apply the config.

right I get locked out here. But my Switch is also using tagged packages here and when I'm connected with Luci and I define a untagged VLAN as Standard I thought that I can still access the Router.
Also when I'm using Wifi and created vlan adapter is bound to my wifi interface. Otherwise I thought wrong and need some assistence here ;-)

systemcrash, 2024-07-16:
You can still define untagged (PVID) - it should be u. But perhaps it shall be tagged in the cpu (eth0).

ffainelli, 2024-07-16:
It most likely needs to be tagged for the CPU port, which means that you would need a br-lan.<VID> interface defined to terminate that VLAN tag at the CPU port level. This is echoing back to https://github.com/openwrt/openwrt/issues/14195#issuecomment-1850517349

palbiez, 2024-07-17:
I tested today on my test device. Dlink DAP1325 MediaTek MT7628NN

Activated the vlan_filtering enabled at CPU 

The test-bridge with vlan was only set to one device. But not to bridge. I will send the configuration later.
Restart network and no further connection available. Not with port 22 and also no other connections are available.
I checked with Wireshark the network packages and after activating the vlan_filtering Tag only IP V6 and arp packages are sent from this device to his Lanport. 
My mistake was that I forgot to activate the vlan monitoring to activate at my windows device. Tomorrow morning I will check again. 
Any ideas? Is this a bug or a misconfiguration?

palbiez, 2024-07-18:
I think this are two parts. One is Luci. The Vlan filtering checkbox is not correspondign with the vlan filtering Option. Another option can be to write in the documentation that VLAN filtering checkbox is also checking the vlan_filtering on CPU level or something like that.
Another one are my issues with my device which can be also a bug but not a luci bug I will reply there in the forum 
https://forum.openwrt.org/t/cudy-wr1300-v3-23-05-3-add-vlans-crashes-network-adapter/204059/5

Neustradamus, 2024-08-25:
To follow this good ticket :)

jow-, 2024-09-01:
Netifd's implicit default for `vlan_filtering` becomes `1` as soon as there's at least `bridge-vlan` section defined, see here:
https://git.openwrt.org/?p=project/netifd.git;a=blob;f=config.c;h=d187152b37ae9a3ac8d46a0d06890db6ae5f2bfb;hb=68c8a4f94cd3cfd654a52cbc8b57c5c9d99640dd#l125

LuCI should reflect that behaviour;
 - Once at least one bridge vlan is declared, the internal implicit default of the checkbox becomes `true`
 - Once at least one bridge vlan is delcared, LuCI auto-checks the checkbox
 - When a checked checkbox with implicit default `true` is saved, the entire option is removed. By design, LuCI only saves options which deviate from the defaults
 - If you get "locked out" after applying such a change it is either because you're doing these while connected wired and your client connection is not updated accordingly (e.g. PC you're browsing LuCI with is not configured for VLAN tagged traffic) - or - the DSA / bridge-vlan implementation of the target device is bugged

zhanko73, 2025-01-02:
I think I have something similar issue:
**WHAT I WANT TO ACHIEVE?**  Add VLAN 732 to bridge called "vlan" (sorry for the name match but this is my case)
**STEPS TO REPRODUCE:**
STEP1: I have a bridge called "vlan" with some ports (lan2, lan3... vlan.20,  vlan.780)  added and vlan filtering enabled. VLANs on the bridge are 20, 780 and 732. I created the device vlan.732 as network device like in case of vlan.20 and vlan.780 
STEP2:  vlan.20 inerface has an IP. Connect to the box via that IP. The key: I will modify the bridge setup via a port which is member of the bridge.
STEP3: Add a new port to the bridge vlan.732 change untagged and setup the primary vlan.
STEP4: DO NOT touch the vlan filtering which is ticked so theoretically it shows vlan filtering enabled.
STEP5: save + save&apply
 **RESULT:**  box become unresponsive.
**Why?** Via the ssh interface (and using another IP to connect) I can see the bridge called vlan is not up:
```
bridge name	bridge id		STP enabled	interfaces
br-lan		7fff.4cadabaedff8	no		lan1
vlan		7fff.000000000000	no
```		

**WORKAROUND that can be painfull if you forget it**
Repeat from  STEP1 till STEP5 but instead of STEP4 untick the vlan filtering (so the checkbox will be empty, aka disable filtering, so not a typo). Observe when you press save, in the same second the box will be ticked automatically. Then press save&apply.
**RESULT WITH WORKAROUND:** 
Box will remain available.
Checked the config:
```
bridge name	bridge id		STP enabled	interfaces
br-lan		7fff.4cadabaedff8	no		lan1
vlan		7fff.4cadabaedff8	no		extsw
							lan2
							lan3
							lan4
							wan
```
**ROOT CAUSE:**
See the two screenshots attached. 
They are exactly the same except the first line. When the connection is lost actually the vlan filtering will be disabled, however that part of graphical user interface remained unchanged so theoretically the checkbox should remain checked.
So maybe the code of user interface does not read the status correctly.

**DEVICE AND OPENWRT INFORMATION:**
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='24.10.0-rc2'
DISTRIB_REVISION='r28161-ea17e958b9'
DISTRIB_TARGET='bcm53xx/generic'
DISTRIB_ARCH='arm_cortex-a9'
DISTRIB_DESCRIPTION='OpenWrt 24.10.0-rc2 r28161-ea17e958b9'
DISTRIB_TAINTS=''

[image]
[image]

zhanko73, 2025-01-02:
> That `vlan_filtering` is removed is normal. The internal default is 0, which is synonymous with the setting being removed.
>
>
> Do you mean you get locked out and the device stops responding? This is [...]

In my case if I connect to the box via another IP where the given interface is not part of the bridge I modified then I can simple disable vlan filtering (uncheck vlan filtering checkbox) which will actually check it then save/save&apply and box will be available again via the non-working IP too. So this is not related to the vlan tagging.