openwrt/luci #7199: luci-mod-network: invalid dnsmasq instance written to config

hmoffatt, 2024-07-18:
Static leases are saved with invalid dnsmasq interface options. The instance is saved by index, not by name, and does not end up in the resultiing dnsmasq config file.

## Steps to reproduce:
1. Go to network -> DHCP and DNS and create a second dnsmasq instance
2. Create a static lease and choose one of the dnsmasq instances, rather than leaving it unspecified/
3. Save and note the proposed config has instance '0', not the name

```
config host
	option name 'testmac'
	list mac '99:88:77:66:55:44'
	option instance '0'
```

4. Apply and note that the static lease is not in the dnsmasq config file.

## Actual behavior:
1.Selected instance is referenced by number, which does not work.

## Expected behavior:
1.Selected instance is referenced by name, which does work.

## Additional Information:

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='23.05.2'
DISTRIB_REVISION='r23630-842932a63d'
DISTRIB_TARGET='mvebu/cortexa53'
DISTRIB_ARCH='aarch64_cortex-a53'
DISTRIB_DESCRIPTION='OpenWrt 23.05.2 r23630-842932a63d'
DISTRIB_TAINTS=''
```

dannil, 2024-07-18:
Would you be able to test 23.05.3? It sounds like this was fixed with https://github.com/openwrt/luci/pull/6579 and cherry-picked to 23.05 with https://github.com/openwrt/luci/commit/26e67f5ef88dbe114cded88d88ec557ebf0c25ea, which is after the 23.05.2 release but is included in 23.05.3.

hmoffatt, 2024-07-18:
That PR does look relevant, but I tried 23.05.3 and it seems unchanged. I also tried adding a new dnsmasq instance with the new version but static leases are still being assigned by index; the generated commands for a static lease were

```
uci del dhcp.cfg34fe63.mac
uci add_list dhcp.cfg34fe63.mac='99:88:77:66:55:44'
uci set dhcp.cfg34fe63.instance='2'
```

and the lease still doesn't end up in the generated dnsmasq config.

hmoffatt, 2024-07-20:
I've tried now with 23.0.5 and I don't see any change unfortunately.

systemcrash, 2024-07-23:
Should be available in the repo tomorrow.

closed 2024-07-23 by systemcrash

hmoffatt, 2024-07-24:
Thanks. I upgraded luci-mod-network:

```
root@router:~# opkg upgrade luci-mod-network
Upgrading luci-mod-network on root from git-24.111.76511-ff6b275 to git-24.205.48215-4f80a61...
Downloading https://downloads.openwrt.org/releases/23.05.4/packages/aarch64_cortex-a53/luci/luci-mod-network_git-24.205.48215-4f80a61_all.ipk
Configuring luci-mod-network.
```

It doesn't seem to be quite right. I see both my instances in the list when I edit a static lease, but if I edit a lease and save it without any changes the instance is lost.

[image]

[image]

systemcrash, 2024-07-24:
I don't observe those results. What you see happening is the mac getting re-written, since it's now (for simplicity) a list, and not a space separated field. All other properties are retained. You might need to toggle instance to another, save it, edit again and toggle back, save to get the result you need the first time.

hmoffatt, 2024-07-24:
> What you see happening is the mac getting re-written

This seems to happen on every single edit now, btw. Even if I save the changes, next edit does exactly the same thing. Unrelated to this issue of course.

systemcrash, 2024-07-24:
Might be related to the mac verify function that runs. One could add a bit more code to prevent a new write.

commit 5ab0cb1f (2024-07-23): luci-mod-network: Correctly assign dnsmasq instance: name not number

Closes #7199
commit 4f80a617 (2024-07-23): luci-mod-network: Correctly assign dnsmasq instance: name not number

Closes #7199

(cherry picked from commit 5ab0cb1f15bcb31015ae8345d302c6d023ba51e2)
commit 73994dcb (2024-07-23): luci-mod-network: Correctly assign dnsmasq instance: name not number

Closes #7199

(cherry picked from commit 5ab0cb1f15bcb31015ae8345d302c6d023ba51e2)