openwrt/luci #7206: prometheus-node-exporter-lua: Add support for ddns-scripts state

aragilar, 2024-07-22:
It would be nice if information from ddns-scripts was served out from prometheus-node-exporter-lua.

As far as I can see, this is accessible from `/var/run/ddns/`, though there may be a better way than scraping those files?

systemcrash, 2024-07-22:
What (state) information specifically?

aragilar, 2024-07-23:
I would suggest as a minimum when the last check ran for each of the DDNS services, and whether the last check succeeded or failed, but you could also include what the IP address and DNS name pair were, what method was used, when the next check should be, and anything else ddns-scripts provides access to.

systemcrash, 2024-07-23:
If the daemon presents prometheus ready scrape endpoint, then this is possible. This is something that first needs to happen in ddns.

closed 2024-07-23 by systemcrash