openwrt/luci #7211: i18n: some translations have degenerated

jjm2473, 2024-07-25:
For example, "Logout" was translated on https://github.com/openwrt/luci/blob/d24e3be252a088d33cb43932886495fe14bf4ae9/modules/luci-base/po/zh_Hans/base.po#L5005-L5007

but it missing recently https://github.com/openwrt/luci/blob/73994dcb74f92d46341c8e5318d3e173a46a137f/modules/luci-base/po/zh_Hans/base.po#L5135-L5137

## Additional Information:
OpenWrt version information from system `/etc/openwrt_release`

```
DISTRIB_ID='iStoreOS'
DISTRIB_RELEASE='22.03.7'
DISTRIB_REVISION='2024072519'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='iStoreOS 22.03.7 2024072519'
DISTRIB_TAINTS='busybox'
```

systemcrash, 2024-07-25:
DISTRIB_ID='iStoreOS'

closed 2024-07-25 by systemcrash

jjm2473, 2024-07-25:
I don't think this bug is related to `DISTRIB_ID='iStoreOS'`, actually it using luci same as `OpenWRT 22.03.7`.

Maybe there is a bug in the i18n sync script, which causes the translation to be lost.